revk/xESP32-Faikout #708: Help: TC2030 (IDC) cable and blank Faikin's from JLPCB

imdos, 2025-04-18:
I have bought some blank Faikin's from JLPCB and this [TC2030 cable with IDC lead](https://nl.aliexpress.com/item/1005007144118184.html).  

I tried to connect to the chips with a regular USB-lead and some dupont wires connected. Then I get an unknown USB device error on Windows. 

And also with a J-Link segger connected as a JTAG. According to this [article](https://www.segger.com/products/debug-probes/j-link/accessories/adapters/6-pin-needle-adapter/)

Also no connection. 

Some more information. The boards are from the 2025-02-04 revision. 
Command lines used:
esptool-v4.7.0-win64\esptool.exe --chip esp32s3 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin

esptool.exe --connect-attempts 25 write_flash 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin
esptool.py v4.7.0
Found 1 serial ports
Serial port COM9
Connecting................................................................................................................................
COM9 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.

revk, 2025-04-18:
Not sure what to suggest, someone else had a similar problem, but I use the same files for ordering from JLC. I use https://www.tag-connect.com/product/tc2030-usb-nl connected to a Mac and have no problems at all.

I'll be interested to hear what the issue is if you work it out.

imdos, 2025-04-19:
Fixed it myself eventually. But I'm afraid I burnt one faikin while trying to get it working. 

PS C:\Users\sande\Downloads\faikin> ..\esptool-win64\esptool.exe --chip esp32s3 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin
esptool.py v4.8.1
Found 0 serial ports

A fatal error occurred: Could not connect to an Espressif device on any of the 0 available serial ports.
PS C:\Users\sande\Downloads\faikin> ..\esptool-win64\esptool.exe --chip esp32s3 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin
esptool.py v4.8.1
Found 1 serial ports
Serial port COM7
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 98:3d:ae:f1:06:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Flash will be erased from 0x00010000 to 0x00170fff...
SHA digest in image updated
Compressed 14064 bytes to 9755...
Wrote 14064 bytes (9755 compressed) at 0x00000000 in 0.2 seconds (effective 462.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 194...
Wrote 3072 bytes (194 compressed) at 0x00008000 in 0.1 seconds (effective 480.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 741.5 kbit/s)...
Hash of data verified.
Compressed 1445888 bytes to 902650...
Wrote 1445888 bytes (902650 compressed) at 0x00010000 in 13.1 seconds (effective 880.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

[image]

imdos, 2025-04-19:
Don't know if there is a way to revive it. But I might have pinned it incorrectly. 

You may add this to the wiki and let me know if you require a better picture.