revk/xESP32-Faikout #715: [BUG] MQTT failed

Exporter8220, 2025-05-12:
Faikin-S3-MINI-N4-R2

The faikin gives the error message MQTT Failed.

It seems that I used the right host/user/password. But it still gives the error message. I use HA. I could not find experiences with the same issue.

Any suggestions?

[image]

revk, 2025-05-13:
8123 is a very unusual port for MQTT, that is normally the web interface for HA. Are you sure you have MQTT on that port?

Also, I don't think we handle MDNS client, so .local domains won't work.

Have you tried just putting the IP address of your home assistant server, no port?

closed 2025-05-13 by revk

Exporter8220, 2025-05-13:
I tried the HA IP before and it gave the error messsage MQTT not connecting. I tried it again and still comes up with the MQTT not connecting error. Discovery is on, on the HA MQTT Server. 

[image]

revk, 2025-05-14:
You have the MQTT module in HA? Non secure MQTT, yes?

Exporter8220, 2025-05-14:
Thank you for pointing me in that direction. Non-secure ports were not filled in. It works now. 

Thank you!

revk, 2025-05-14:
Good. If you want to use secure ports you need to set the mqtt certificate in the library settings on the Faikin. But insecure ports are usually OK on a local network.