revk/xESP32-Faikout #716: [BUG] System is offline (no data tx/rx)

BlackEdder0815, 2025-05-13:
**Faikin hardware**
Newly bought Faikin-Board by Amazon (3x)
Faikin-S3-MINI-N4-R2: 8ab9fe5 2025-05-05T13:30:30

**Daikin hardware**
CTXM15R2V1B (2022)
Connected directly (1:1 wiring) with jumper-cables. I measured the cables due to checking, if the wiring is correct and without resistance/interruptions.

Original wifi-controller
[image]

Connector on mother board
[image]

**Describe the bug**
Faikin is powering up, but no connection. Frontend shows:
System is offline (no data tx/rx)
mqtt-topics:
status:
```
{
  "protocol": "S21",
  "online": false,
  "autor": 0,
  "autot": 0,
  "auto0": "00:00",
  "auto1": "00:00",
  "autop": false,
  "autoe": true
}
```
comms:
```
{
  "protocol": "Altherma_S",
  "payload": "",
  "bad-cs": "FF",
  "bad-cmd": "U"
}
```
tx:
```
{
  "protocol": "Altherma_S",
  "cmd": "S",
  "dump": "0253AA"
}
```
cannot-tx:
```
{
  "protocol": "X50A¬Tx¬Rx",
  "cmd": "CB",
  "payload": "0000"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connecting/wiring
2. powering up

**Expected behavior**
Getting a connection

**Additional context**
I read already a lot, but didn't found a valid solution, yet.
I checked another Faikin-Board (ordered 3) and another Daikin-device (I have 3). But no difference.
Any advice would be helpful, why this is not working. https://github.com/revk/ESP32-Faikin/wiki/List-of-confirmed-working-air-con-units lists my model (most likely)  
[image]

I used the original cable, unpinned the connector on one side and used jumer-cables (male-male) to establish the connection (one side, simply put the unpinned pin into the jumper-cable, other side directly to the Faikin-Board).

Any ideas, where to start with debugging?

revk, 2025-05-13:
If you loop back the tx and rx on the Faikin module and power it, does it say it is in loopback? LED cycles colours.

BlackEdder0815, 2025-05-13:
I have this already done more than once. I also changed the cable, and measured the connection from the Faikin-device to the end of the connector. 
Does it help to connect an oscilloscope? What should I measure on to or rx?

revk, 2025-05-13:
Only other possible is the 5V line not bing 5V to pull up or something like that maybe, but even that is a stretch.

Someone else with the same Dakin air-con may be able to advise perhaps. Loopback proved the Faikin is working, as I say, but maybe there is something else to it.

Worth raising as a "discussion" maybe?

BlackEdder0815, 2025-05-15:
Alright, maybe I got it.
I was on the original WIFI-socket, what's obviously wrong. The S21 is above this socket (and not connected by default).
I thought, I have to replace the original WIFI module, as this is also connected to the S21.
I have to wait for a matching connector cable. I will confirm, if it's working then. If yes, I will close this issue. Thanks for your support.

BlackEdder0815, 2025-05-17:
Unfortunately no change. I fear, the protocol has changed.

BlackEdder0815, 2025-05-30:
At least the cause for the initial intend to create this issue is gone. 
See this [discussion](https://github.com/revk/ESP32-Faikin/discussions/718) for the resolution. Thanks for your contributions!

closed 2025-05-30 by BlackEdder0815