revk/xESP32-Faikout #719: [FEATURE]

ernesto75, 2025-05-16:
**Daikin hardware**
FTXF25E5V1B with BRP069C47 

**Is your feature request related to a problem? Please describe.**
I'd like to know if I can get rid of the BRP069C47 wifi module and replace it with a faikin. 

**Describe the solution you'd like**
Replace current module with faikine in the same spot and with same cabling. But my current module is attached to PCB port S430. I'd like to know if is compatible with faikin. I attach picture of schematics at bottom section.

**Describe alternatives you've considered**
Best alternative would be using a BRP069B41 but seems not to be compatible with my ac unit.

**Additional context**

[image]
[image]

MUN0X, 2025-05-16:
Your unit has a s403 port.
[see here](https://github.com/revk/ESP32-Faikin/issues/208)

pplucky, 2025-08-13:
Hi there.

I have a similar case as yours and was wondering if you ever got to any conclusion.

I was wondering if the end cable that connects to the BRP069C47 board (named Adapter side connector) would somehow be compatible with Faikin, as I would rather NOT play around too much with the S403 connector (which my unit also has).

Thanks in advance