revk/xESP32-Faikout #720: Control upper/lower airflow direction on Daikin floor unit

elindse, 2025-05-16:
Hi, and thanks for your work on Faikin

I’m using a Daikin floor-mounted heat pump unit (model FVXM35A + RXTP35R) with dual airflow outlets—one at the top and one near the floor. This model supports directing air through either the upper or lower outlet, or both simultaneously.

Using the official Daikin controller, I can manually switch between:
- Upper outlet only
- Lower outlet only
- Both outlets

However, this functionality does not currently appear to be exposed in the Faikin integration.

Feature request:
Please consider adding support for controlling the airflow outlet mode (top, bottom, or both) to the Faikin.

Use case:

During heating, I want to prioritize lower airflow (along the floor).

During cooling, I want to prioritize upper airflow (to avoid cold air along the floor). 

Hardware details:

Indoor unit: Daikin FVXM35A

Outdoor unit: RXTP35R

revk, 2025-05-16:
That sounds cool.

We'd need to know the S21 messages to do that though, and I don't know if anyone has found commands for that yet.

MUN0X, 2025-05-18:
You can control it via the IR remote according to the manual.
Change it and monitor s21 messages for canges with faikin

sam-m7, 2025-07-11:
A while ago I tried to find the S21 command for using only the floor outlet while heating (floor warming mode), checking if something changes in the S21 messages. I couldn't find something unfortunately. 

I wasn't aware you can close the lower lid in cooling mode though (through the remote or the official app). I only noticed the AC controlling that automatically. If it's possible, it might be related to #759. My last reply there might be incorrect then.

MicF1968, 2025-10-03:
Same issue here.
The floor standing models control the bottom flap via the vertical swing.
Instead of just having vertical swing on or off, these models have vertical swing states: off, swing, bottom

[image]

I have no idea to inspect mqtt settings, though.