revk/xESP32-Faikout #721: Config Daikin from a Win PC/Laptop

GerhardKreuzer, 2025-05-18:
Dear Team,
I tried to config a Faikin modul. Connection to Daikin split is ok, Led flasching green/red.
I don't have an IPhone, so the setup.md is of less help for me.
With ipconfig I get the address of the module 10.187.228.1, it get the addreess of an gateway, ok ... why not.
I entered this IP into the browser an get a web page, asking me for the WiFi settings.
I entered the SSID and Pwd and hit 'Save' and get a strange message.
Now the Led is blinking as before, the Faikin-xxx network is gone and the module is not in my network.

PLEASE, can I set a static IP (cause the module is part of a split box mounted on a wall and so we can call it static)?

How can I find out whats happen now and how can I proceed?

With best regards

Gerhard

revk, 2025-05-18:
If it is not longer offering an AP/SSID then it is on your network, you just need to find the IP. It is meant to display than once you enter the details.

Once set up, you can set a static IP, but that is not shown on that initial page, it assumes it can get an IP by DHCP.

Once set up and given a name it should be accessible using MDNS, i.e. itsname.local

Your DHCP server should be able to tell you.

Otherwise you can factory reset - the two round pads on back - short once and LED goes yellow, twice, and then third time it goes red and resets back to factory defaults and shows as an SSID/AP again.

GerhardKreuzer, 2025-05-18:
Sorry, but this module do its own ..
I connected the pads, green, yellow, red. I see the Faikin-xxx network again. I entered the same network data, checked it more than one time, proof everything is correct. Push 'Save'

[image]

Is this correct?

I disconnected from the Faikin-xxx network and uses my LanMonitor to find the module in the DHCP list, no module. I am still able to connect via Faikin-xxx.

Sorry but there is something different.

After a power cycle the Faikin-xxx net is gone but the module isn't in my net.

Led still blinking green mixed with some other color.

Any chance to find out whats going on?

Can I try the procedure without air condition connected (in my electronic lab) just to see, how it really works?

With best regards

Gerhard

revk, 2025-05-18:
Just to check, you have 2.4GHz wifi, yes?

When you save, what comes up on the screen at that point? It *should* say connected and an IP.

GerhardKreuzer, 2025-05-19:
WiFi is 2.4Ghz.
Maybe the signal is too weak. Today the module appears in the LanMonitor, but I was not able to ping.
Can I test a module without AC unit?
Problem is to disassemble the AC units which is lot of work and than reassemble it because the clients coming through on weekdays, the other unit is in an empty room so I can leave it as is, but WiFi ... With my laptop Wifi is ok.

With best regards
Gerhard

revk, 2025-05-19:
WiFi and web settings should all work standalone. Yes.

GerhardKreuzer, 2025-05-19:
Thanks. Looks like I am there only next sunday evening ...