revk/xESP32-Faikout #728: [BUG] S21/S403 Retrying, Mismatch

vinicius-trev, 2025-05-28:
**Faikin hardware**
ESP32S3-DevKitC-N8R8 + BSS138 Level Shifter + 12V to 5V buck converter

**Daikin hardware**
CTXC12RMVM - S21/S403

**Describe the bug**
What is the best way to debug this error:
```
comms = {"protocol":"S21","mismatch":true,"data":"02473730304E4A7603"}
```

The behavior that I am having is quite strange, powering on/off, setting the temperature, econo/powerful/comfort/quiet modes and vertical swing work flawlessly but still gives me that mismatch message. On the other hand trying to change the mode or fan makes the unit beep 3-6x giving that mismatch message too... I can see that the unit is changing its settings, but the firmware is probably retrying to send the settings over and over again.

**Troubleshoot**
1. Already tried to replace the BSS138 level shifter with another one, same results
2. Latest Beta FW `dc955d0 2025-05-23T11:03:02`
3. X50A, CN_WIRED and Swap RX/TX disabled
4. All tests done using the built-in frontend

Dump files from MQTTX for detailed troubleshooting:
1. [mqtt_turnon_idle.json](https://github.com/user-attachments/files/20473374/mqtt_turnon_idle.json): Just powering on, without changing any settings (success, at least based on feedback from the unit itself)
2. [mqtt_set_econo_comfort.json](https://github.com/user-attachments/files/20473384/mqtt_set_econo_comfort.json): Setting `Econo mode` and `Comfort mode` (success, at least based on feedback from the unit itself)
3. [mqtt_fan_auto_5_error.json](https://github.com/user-attachments/files/20473490/mqtt_fan_auto_5_error.json): Changing fan speed from `Auto` to `5` (here is where the Faikin seems to be retrying to send the command, given that the unit keeps beeping for 3-6x (random, and sometimes only once) even in real-life noticing that its behavior changed as the desired settings. This also happens when trying to change the working mode

vinicius-trev, 2025-05-28:
At the HA UI, when I change the operation mode or fan speed for example, it tries 2-6x as you said, even working on the first try and at the end it of those tries it persists the settings. I don't know either what is happening... quite odd, I may do more investigation in it I just need more directions on what to do.

Is there a possibility that both internal units are communicating themselves?

revk, 2025-05-28:
Seems unlikely. There are `debug` and `dump` settings to cover everything we send and receive. There is some work on S21 in some of the discussions here as well.

vinicius-trev, 2025-05-28:
Is there a way to change the pooling rate or the S21 or timeout operations? How does the code pools for the updated setting after changing it?

I will try to investigate the error case using dump and debug in more details, any ideas where to start there?

vinicius-trev, 2025-05-28:
Some updates, for every command that I use, I kind of receive a mismatch, even if it works in the unit.

When sending a F1/D1 cmd with: `On, Fan Mode, Night, 18C`. I get a `16@B`, which is correct, for both TX and RX with the Ack

With that, I have no clue why it is retrying and receiving a mismatch.

[image]

revk, 2025-05-28:
`F5` is part of normal polling expecting a `G5` response.

The duplicate from air-con is the issue. Is this the latest beta?

vinicius-trev, 2025-05-28:
Yes, it is, version: 9a11b89 2025-05-28T15:06:47

At least the UI is showing as up to date.

revk, 2025-05-28:
OK try beta again now please.

vinicius-trev, 2025-05-28:
This is an example that shows the data is not bein replicated when trying to enable `Econo Mode` 
```
{tx} {"protocol":"S21","dump":"024437303230303D03","D7":"0200"}
{rx} {"protocol":"S21","cmd":"D7","payload":"30323030","text":"0200","ack":true}
{tx} {"protocol":"S21","dump":"0246317703","F1":""} 
{rx} {"protocol":"S21","dump":"02473730304E537F03","G7":"00NS"}
{co} {"protocol":"S21","cmd":"F1","mismatch":true,"data":"02473730304E537F03"}
```

Do we need to skip sending the F1 in this case? Is the hvac replying the G7 msg given the first D7? 

I will try the new beta in a couple minutes.

revk, 2025-05-28:
I don't know what it is doing, but see if we have any progress on latest.

vinicius-trev, 2025-05-28:
Version: d74174e 2025-05-28T16:01:36 S21, same behavior.

### Test 1
```bash
# Toggling Vertical Swing
M1 tx {"protocol":"S21","dump":"024435313F30304903","D5":"1?00"}
M2 rx {"protocol":"S21","cmd":"D5","payload":"313F3030","text":"1?00","ack":true}
M3 tx {"protocol":"S21","dump":"0246317703","F1":""}
M4 rx {"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"} 
M5 co {"protocol":"S21","cmd":"F1","mismatch":true,"data":"024735303030808C03"}
M6 rx {"protocol":"S21","dump":"02473131364E416E03","G1":"16NA"}
M7 tx {"protocol":"S21","dump":"0246317703","F1":""}
M8 rx {"protocol":"S21","dump":"02473131364E416E03","G1":"16NA"}
```
As far as I could see in this test, I am getting an [M4 for the M1] and an [M6 for the M3] requests, but they are out of order

### Test 2

```bash
# Changing Fan Speed
M1 tx '{"protocol":"S21","dump":"024431313640425E03","D1":"16@B"}'
M2 rx '{"protocol":"S21","cmd":"D1","payload":"31364042","text":"16@B","ack":true}'
M3 tx '{"protocol":"S21","dump":"0246317703","F1":""}'
M4 rx '{"protocol":"S21","dump":"02473131364E416E03","G1":"16NA"}'
M5 tx '{"protocol":"S21","dump":"0246357B03","F5":""}'
M6 rx '{"protocol":"S21","dump":"02473131364E416E03","G1":"16NA"}'
M7 co '{"protocol":"S21","cmd":"F5","mismatch":true,"data":"02473131364E416E03"}'
M8 rx '{"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"}'
```

In this one, on the other hand, I am getting an [M4 for the M1], [M6 for the M3] and [M8 for the M5] requests, but they are out of order too. 

Those are just assumptions looking at the data, I may be wrong.

revk, 2025-05-28:
OK, looking at the `D1`, it has `16@3` which is `1` (on), `6` (mode fan), `@` (temp 18.0C), `3` (fan level 1)

The `F1` poll first gets `16U7` which is same but temp 28.5C and fan level 5.

The next, and later`F1` poll gets `16U3`, which is the same except temp 28.5C.

So sounds like the issue is the temp being sent. But the fan is being confirmed as `3`.

So the resends do not look fan related, they look temperature related, but we ignore temp in *fan* mode, so that is odd.

Now there is a second way we get fan mode though. Do you see any examples if poll `RG` and `SG` response, if so the *fan* mode is recognised from that.

vinicius-trev, 2025-05-28:
Yes:
```
'{"protocol":"S21","dump":"0252479903","RG":""}'
'{"protocol":"S21","dump":"02534737D103","SG":"7"}'

(...)
'{"protocol":"S21","dump":"0252479903","RG":""}'
'{"protocol":"S21","dump":"02534733CD03","SG":"3"}'
```

I am doing those tests in Fan Mode just to test, but I can do them at the Cool mode and will probably get the same results

revk, 2025-05-28:
OK and `SG` of `3` is fan mode 1, which should complete the command.

Does it log a `fail-set` message?

vinicius-trev, 2025-05-28:
> Does it log a `failed-set` message?

No, nothing at `failed-set`.

revk, 2025-05-28:
OK, that makes no sense. It sends until polling confirms the response and should then stop sending. If it tried too many times it gives up and lots `failed-set`.

Only think I can think is I log when it *sees* the response match what it wants as a special extra message.

That may have to wait until tomorrow now, sorry.

vinicius-trev, 2025-05-28:
### Cool Mode
#### Changing fan speed from Low (3) to High (7)
```
#[Failed] First try
'{"protocol":"S21","dump":"0246317703","F1":""}'
'{"protocol":"S21","dump":"024731313355336403","G1":"13U3"}'
'{"protocol":"S21","dump":"0246357B03","F5":""}'
'{"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"}'
'{"protocol":"S21","dump":"0246367C03","F6":""}'
'{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}'
'{"protocol":"S21","dump":"0246377D03","F7":""}'
'{"protocol":"S21","dump":"024737303056548803","G7":"00VT"}'
'{"protocol":"S21","dump":"02464D9303","FM":""}'
'{"protocol":"S21","dump":"02474D373930306403","GM":"7900"}'
'{"protocol":"S21","dump":"0252489A03","RH":""}'
'{"protocol":"S21","dump":"0253483538322B6503","SH":"582+"}'
'{"protocol":"S21","dump":"024431313355376503","D1":"13U7"}'
'{"protocol":"S21","cmd":"D1","payload":"31335537","text":"13U7","ack":true}'
'{"protocol":"S21","dump":"0246317703","F1":""}'
'{"protocol":"S21","dump":"024731313355336403","G1":"13U3"}'
'{"protocol":"S21","dump":"0246357B03","F5":""}'
'{"protocol":"S21","dump":"024731313355376803","G1":"13U7"}'
'{"protocol":"S21","cmd":"F5","mismatch":true,"data":"024731313355376803"}'
'{"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"}'

#[Failed] Second try
'{"protocol":"S21","dump":"0246317703","F1":""}'
'{"protocol":"S21","dump":"024731313355376803","G1":"13U7"}'
'{"protocol":"S21","dump":"0246357B03","F5":""}'
'{"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"}'
'{"protocol":"S21","dump":"0246367C03","F6":""}'
'{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}'
'{"protocol":"S21","dump":"0246377D03","F7":""}'
'{"protocol":"S21","dump":"024737303056548803","G7":"00VT"}'
'{"protocol":"S21","dump":"02464D9303","FM":""}'
'{"protocol":"S21","dump":"02474D373930306403","GM":"7900"}'
'{"protocol":"S21","dump":"025261B303","Ra":""}'
'{"protocol":"S21","dump":"0253613530332B7703","Sa":"503+"}'
'{"protocol":"S21","dump":"024431313355376503","D1":"13U7"}'
'{"protocol":"S21","cmd":"D1","payload":"31335537","text":"13U7","ack":true}'
'{"protocol":"S21","dump":"0246317703","F1":""}'
'{"protocol":"S21","dump":"024731313355376803","G1":"13U7"}'
'{"protocol":"S21","dump":"0246357B03","F5":""}'
'{"protocol":"S21","dump":"024731313355376803","G1":"13U7"}'
'{"protocol":"S21","cmd":"F5","mismatch":true,"data":"024731313355376803"}'
'{"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"}'

#[Success] Third try
'{"protocol":"S21","dump":"0246317703","F1":""}'
'{"protocol":"S21","dump":"024431313355376503","D1":"13U7"}'
'{"protocol":"S21","cmd":"D1","payload":"31335537","text":"13U7","ack":true}'
'{"protocol":"S21","dump":"0246317703","F1":""}'
'{"protocol":"S21","dump":"024731313355376803","G1":"13U7"}'
'{"protocol":"S21","dump":"0246357B03","F5":""}'
'{"protocol":"S21","dump":"024731313355376803","G1":"13U7"}'
'{"protocol":"S21","cmd":"F5","mismatch":true,"data":"024731313355376803"}'
'{"protocol":"S21","dump":"024735303030808C03","G5":"000\\u0080"}'
```

#### Changing fan speed from High (7) to Low (3)
```
```

revk, 2025-05-28:
Ok I have worked it out.

Because your device does the `RG`/`SG` command, we use that for fan setting confirmation, and not `F1`. This is because `F1` does not do *quite* mode.

But the `R` polls are slower, so it ends up taking many poll cycles, trying to set send *fan* to before it does the `RG` poll.

Let me adjust that!

vinicius-trev, 2025-05-29:
BTW, we can mark the indoor unit CTXC12RMVM as partially compatible with the Faikin/S21/S403, I tested it all day and only noticed that double beep while changing operation mode. Swing at the HA also doesn´t work if I disable the horizontal, which my unit is not capable of; all the rest are working I can debug those later.

Maybe in the future, add checks to not enable powerful, econo and comfort while the unit is off, but those are nitpicks

For those with similar issues in that same unit, also make sure to not enable the `nodemand`, as it leads to an unoperational unit.