revk/xESP32-Faikout #735: [BUG] Powerful mode doesn't tun ON/OFF through webUI

baloo2, 2025-06-05:
**Faikin hardware**
Faikin/ESP32 purchased from Tindie

**Daikin hardware**
FTXG50JV1BW + S21

**Describe the bug**
I cannot turn ON/OFF Powerful Mode. If turned ON through UI, it reverts back to OFF. UI doesn't reflect the change if turned on by IR remote controller. 

**To Reproduce**
Turn On Powerful mode in UI or Home assistant.

**Expected behavior**
Daikin should reflect changes of Powerful mode  in UI/HA.

**Additional context**
I bought Faikin in 2/2025 and powerful mode worked with that version of FW. Since then there were couple of updates so it has to be caused by one of them (I started using AC only this week). Screenshots from MQTT Explorer are attached (shouldn't state of powerfull be true or false only?)

[image]
[image]

revk, 2025-06-05:
That usually means the S21 message to do *powerful* is not working, not all units handle all S21 messages.

baloo2, 2025-06-05:
As I wrote above, **Powerful mode have been working** with exact same HW setup. It just got broken with subsequent revK updates. That was the reason I tag it as BUG.

revk, 2025-06-05:
OK, sorry, can you turn on the `dump` setting and get MQTT messages for a change, it should like what it sends and what we get back in the S21 messages.

revk, 2025-06-05:
Just to confirm, testing here on latest beta, powerful is working, so I cannot reproduce this, hence needing the message dumps.

revk, 2025-06-05:
I have no idea how you export - but simple topic and payload one per line is a good idea.

Also, worth trying beta release anyway in case this is already fixed.

baloo2, 2025-06-05:
Installed newest Beta - powerful doesn't work as well. Will try to export dumps for you.

baloo2, 2025-06-05:
Here is dump (applied 2 times Powerfull):
[dump.txt](https://github.com/user-attachments/files/20609982/dump.txt)

revk, 2025-06-05:
OK good news is it logs sending a `D3` with `0002`, which is *powerful* on.

It also logs polls with `F3`, with `G3` response `000\u000` (i.e. three zeros and a null) response. The problem is that is a *powerful* not on, hence the `failed-set`.

Can you log just after a reboot looking for `F6` at all? I suspect they fail, hence the `D3` and `F3` logic, but worth double checking.

baloo2, 2025-06-05:
Here is dump after restart + 1x powerfull applied:

[dump_restart.txt](https://github.com/user-attachments/files/20610205/dump_restart.txt)

baloo2, 2025-06-05:
Just a note: after restart Powerfull button is not present in webUI - I have to refresh page to appear. This is definitely not like it used to be at the beginning (when I installed Faikin with original firmware).

revk, 2025-06-05:
OK so `F6` gets no reply, as expected. `F3` does, and it is *powerful* off.

So a tad at a loss here. `D3` to set *powerful* seems to simply be not working. What we don't have is a dump from older code when it did work, apparently.

baloo2, 2025-06-05:
I wrote to Elfatronics if they can send me original firmware they shipped the board with. So nothing else can be done from your side?

revk, 2025-06-05:
If you can set up on a URL it can upgrade from that (setting `otahost`) otherwise via USB.

revk, 2025-06-18:
The problem is we are sending S21 messages.

And I only have one Daikin with S21 and it works. So basically needs someone to experiment with S21 on that model.

That is, if that model will do it via S21, which is not necessarily the case.

bittylicious, 2025-07-29:
I'm also generally available to help if you like. Powerful via Faikin doesn't work on my FTXS35D3VMW.

Is anything emitted on the S21 interface when the remote control changes modes etc? The remote allows me to set powerful of course.

asund, 2026-01-19:
@bittylicious https://github.com/revk/ESP32-Faikout/issues/556 is tracking making powerful feedback (not control) visible.