revk/xESP32-Faikout #738: [FEATURE] MQTT temperature sensor implementation

baloo2, 2025-06-07:
Hi, first of all thank you for your great project - I recently started to use Auto mode with my Daikin AC and it is great. Currently I use integrated temperature sensor in Daikin unit with it.

I know there is implementation of external Bluetooth temperature sensor in Faikin however most of us use Faikin together with Home assistant so we have a lot of temperature sensors that are available through MQTT (in my case with Zigbee2MQTT).

As you have already implemented MQTT stack in Faikin, I think it can be great if Faikin can subscribe to some topic that includes temperature sensor. In that case I don't have to use bluetooth sensor but instead can use already  available MQTT temp sensor.

What do you thin about it?

baloo2, 2025-06-07:
Thank you - I found the control message, but you have to actively send the control message.

What I mean is Faikin just to subscribe to MQTT topic on the same broker Faikin is connected to (for example "zigbee2mqtt/Sensor 3"), read "temperature" value from that payload and update the environment temperature the same way as you do it with external bluetooth sensor.

revk, 2025-06-07:
Not impossible, obviously a lot more fiddly as it will be a topic and probably assuming JSON payload and picking a value from that. I'll have to have a think.

baloo2, 2025-06-07:
Thanks. This for example standard payload from temperature sensor in zigbee2mqtt (which I think a lot of people use):

zigbee2mqtt/Sensor 1:
`{"battery":100,"humidity":52,"linkquality":58,"temperature":26.2,"update":{"installed_version":8448,"latest_version":8448,"state":"idle"}}`

revk, 2025-06-07:
OK good that it is JSON. It seems to be becoming de facto standard for MQTT payloads these days.

I'll look in to it.

baloo2, 2025-06-07:
BTW how often control message with env temperature should be sent to Faikin (so it is correctly retained)?

revk, 2025-06-07:
I think it is every few minutes at the longest, so I also need to consider how that will apply in this sort of case.

revk, 2025-06-07:
OK try latest beta, it has `autotopic` and `autopayload` now.

e.g. set `autotopic` to `zigbee2mqtt/Sensor 1`, and `autopayload` to `temperature`

closed 2025-06-07 by revk

baloo2, 2025-06-07:
That was ultrafast implementation. Thank you very much - it works. I will test it and let you know in case of any issues.

Just to clarify - env temp is used only if auto mode is ON (otherwise internal AC temp sensor us used), right?