revk/xESP32-Faikout #740: [BUG]FTXM28VVLT / RXM28VVLT 冷氣的 Streamer, Sensor mode, Quiet outdoor, Econo mode 功能無法運作 (S21 協定)

Brian8901, 2025-06-09:
Faikin hardware (Faikin 硬體)

這是一個 ESP32-S3-MINI-1 模組。我沒有從 Amazon 購買完整的 Faikin 產品，而是自行組裝或購買模組。

板子的正面照片請參考附件。背面照片目前沒有提供。

Daikin hardware (大金冷氣硬體)

我的大金冷氣機型號是：FTXM28VVLT / RXM28VVLT (大金橫綱V 系列)。
我的 Faikin 模組是透過 S21 專用線連接到冷氣機的。

Describe the bug (錯誤描述)

我的大金冷氣機支援「Streamer（閃流放電）」、「Sensor mode（感應模式）」、「Quiet outdoor（室外機靜音）」和「Econo mode（省電模式）」這些功能。

問題是，無論是在 Home Assistant 的控制介面，還是直接在 ESP32-Faikin 模組本身的網頁控制介面 (例如 http://192.168.6.161) 上操作這些功能的開關時，冷氣機都沒有任何反應。目前只有冷氣的「Power」（電源開關）功能是正常運作的。

To Reproduce (重現步驟)

請按照以下步驟操作以重現問題：

開啟 ESP32-Faikin 的網頁控制介面 (例如 http://192.168.6.161)。
點擊「Streamer/filter」、「Sensor mode」、「Quiet outdoor」或「Econo mode」功能的開關。
觀察冷氣機，它不會對這些指令做出任何反應。
Expected behavior (預期行為)

我期望當我在 ESP32-Faikin 的網頁控制介面（或 Home Assistant 介面）上點擊「Streamer」、「Sensor mode」、「Quiet outdoor」和「Econo mode」功能的開關時，我的大金冷氣機應該能正常啟動或關閉這些功能。

Additional context (其他補充資訊)

我的大金冷氣機型號 FTXM28VVLT / RXM28VVLT 確定支援上述所有功能。
ESP32-Faikin 的韌體版本是：Faikin-S3-MINI-R2 8ab9fe5 2025-05-07T13:30:30 S21。
在 ESP32-Faikin 的設定頁面中 (Extra 分頁)，我已經確認了所有相關的「不支援此功能」（例如 nostreamer、nosensormode、noquiet）的開關都是**關閉（OFF）**的，這表示模組應該嘗試支援這些功能。
在 Home Assistant 的整合介面中，這些功能也顯示為關閉，並且無法正常操作。

[image]

[image]

[image]
[image]
[image]
[image]

revk, 2025-06-09:
Sorry, can we have bug reports in English please.

Brian8901, 2025-06-11:
{
"ts":"2025-06-11T01:46:02Z",
"id":"24587CDB7B78",
"up":true,
"uptime":1284,
"mqtt-up":1280,
"app":"Faikin",
"version":"8ab9fe5",
"build-suffix":"-S3-MINI-N4-R2",
"build":"2025-05-05T13:30:30",
"flash":4194304,
"rst":3,
"mem":156644,
"spi":2090980,
"ssid":"Unifi-70",
"bssid":"602232E59BDE",
"rssi":-27,
"chan":11,
"ipv4":"192.168.6.161",
"online":true,
"power":true,
"target":28.00,
"actarget":28.00,
"temp":26.00,
"achome":26.00,
"outside":34.50,
"liquid":14.50,
"demand":100,
"Wh":1213500,
"fanfreq":580,
"comp":960,
"mode":"cool",
"action":"cooling",
"fan":"auto",
"streamer":true,
"quiet":true,
"econo":true,
"comfort":false,
"powerful":false,
"sensor":true,
"swing":"off",
"preset":"eco",
"autoe":false
}

Brian8901, 2025-06-11:
"Here are the dump and error logs when I tried to toggle the Streamer, Sensor, and Quiet modes. As you can see, the failed-set message indicates that setting these features failed, even though the tx message was sent."

{"protocol":"S21","dump":"024742303030304903","GB":"0000"}
{"protocol":"S21","dump":"0246377D03","F7":""}

{"G1":"31335541","G2":"3C3B0080","G3":"30FEFE00","G4":"30008030","G5":"30303080","G6":"30303030","G7":"30303030","GA":"31383630","GB":"30303030","GG":"30453031","GK":"313B3D31","GM":"38374632","GN":"30303030","GP":"45323030","GQ":"45333030","GS":"30303030","GT":"31303030","SD":"303030"}

{"protocol":"S21","ts":"2025-06-11T05:31:00Z","online":true,"home":26.00,"heat":false,"fanrpm":580,"comp":16,"outside":[34.00,34.45,34.50],"liquid":[15.00,15.10,15.50],"anglev":110,"Wh":1215200,"power":true,"mode":"C","temp":28.50,"demand":100,"fan":"A","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":false,"sensor":false,"led":true,"quiet":false}

{"protocol":"S21","cmd":"D6","payload":"B0B0303C","text":"\u00B0\u00B00<","nak":true}

{"streamer":true,"sensor":true,"quiet":true}

revk, 2025-06-11:
The *nak* to the `D6` means the air-con rejected that command.

`D6` is one of the S21 commands to change *powerful*, *streamer*, *comfort*.

Normally if `D6` is not working we fall back to using `D3`.

Are you on latest software? Maybe try beta release?

Brian8901, 2025-06-11:
"I have updated to the beta release firmware (b7ce4f0). However, the issue persists. The air conditioner is still sending nak to the D6 command, and the failed-set error is still being reported for Streamer, Sensor, and Quiet modes.

Here are the latest logs:

Faikin-S3-MINI-N4-R2: b7ce4f0 2025-06-07T13:25:56 S21

{"protocol":"S21","cmd":"D6","payload":"B0B0303C","text":"\u00B0\u00B00<","nak":true}

{"streamer":true,"sensor":true,"quiet":true}

revk, 2025-06-11:
OK the logic is we use `D6` if the `F6` poll is working, and the `D3` command if the `F3` poll is working.

But that is not helpful as `D3` only contains the *powerful* setting.

So basically we are right to use `D6` if your a/c has *streamer*, *sensor*, *comfort*, *quiet*, and *powerful*.

I don't know which your model actually supports, it may be that the NAK is because we are trying one of the settings it does not understand.

You can try various options to adjust what is sent in `D6`, including `noled`, `nosensor`, `nocomfort`, `noquiet`, `nopoweful`, and see if this stops the NAK happening.

closed 2025-06-11 by revk

Brian8901, 2025-06-11:
I've used my Broadlink RM4C Pro to capture the IR codes from my Daikin remote for the functions that are currently not controllable via Faikin. I hope these provide some clues for the S21 protocol.

Here are the captured IR codes:

{
    "Daikin": {
        "Quiet outdoor": "JgCKAg0NDw4ODg8NDw4PAAM2cjgPKg8ODg4PDQ8qDw4ODg8NDw4OKg8ODyoOKg8ODykPKg8qDyoPKg4ODw0PKg8ODg4PDQ8ODg4PDQ8ODw0PDg4ODw4OKg8ODg4PDQ8ODg4PDQ8qDw4ODg8qDioPKg8qDyoPKg4qDw4ODg8NDyoPKg8NDw4PDQ8ODg4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg8NDw4OKg8ODioPKg8qDyoPDQ8ODg4PDQ8ODg4PDg4ODg4PDg4ODw0PDg4ODg4ODg8NDw4ODg8qDw0PDg4ODw0PDg4ODg4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODioPDg8qDg4PDQ8ODw0PDg4ODw0PDg4ODw4ODg8NDw4PDQ8NDw4PDQ8ODg4PDQ8ODg4PDg4ODw0PDg4ODw0PDg8NDw4ODg8NDyoPKg8qDioPKg8qDyoOKg8ABItzOA4rDg4PDQ8ODioPDg8NDw4ODg8qDg4PKg8qDg4PKg8qDioPKg8qDw0PDg4qDw4PDQ8ODg4PDQ8ODg4PDg4ODw0PDg4ODw0PDg4ODw4ODg8NDyoPDg4ODyoOKg8qDw4ODg8NDw4ODg8qDyoOKg8ODg4PDg4ODw0PDg4ODw0PDg8NDw4ODg8NDw4ODg8qDw0PKg8NDw4PDQ8ODg4PDQ8ODg4PDg4ODw0PDg4ODw0PDg4ODw4OKg8qDw4ODg8NDw4ODg8NDw4ODg8ODg4PKg4qDw4PDQ8ODg4PDQ8ODisODg8NDw4ODg8ODg4PDQ8ODg4PDQ8qDw4OKg8ODw0PDg8pDyoPDQ8qDw4ODg8NDw4ODg8qDw0PDg4ODyoPDQ8ODg4PDQ8ODw0PDQ8qDyoPKg8qDioPAA0F",
        "Econo mode": "JgCKAg0NDw4ODg8NDw4OAAM3cjgPKg8NDw4PDQ8qDw0PDg8NDw4OKg8ODioPKg8ODioPKg8qDyoOKg8ODw0PKg8NDw4PDQ8ODg4PDQ8ODg4PDQ8ODw0PKg8NDw4PDQ8NDw4PDQ8qDyoPDQ8qDyoPKQ8qDyoPKg8qDg4PDQ8ODykPKg8ODg4PDQ8ODw0PDg4ODw0PDg8NDw0PDg8NDw4ODg8NDw4PKg4ODyoPKQ8qDyoPDQ8ODw0PDg4ODw0PDg8NDw0PDg8NDw4ODg8NDw4PDQ8ODg4PDQ8qDw4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODw0PDQ8ODw0PDg4ODyoPDQ8rDg0PDg8NDw0PDg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg4ODw0PDg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODioPDg8NDw4ODg8NDw4ODg8ABItyOBApDw0PDg8NDyoPDQ8ODw0PDg4qDw4PKQ8qDw4OKg8qDyoPKg8qDg4PDQ8qDw0PDg8NDw4ODg8NDw4PDQ8NDw4PDQ8ODg4PDQ8ODw0PDg4ODyoPDQ8NDyoPKg8qDw0PDg4ODw0PDg8pDyoPKg8NDw4PDQ8ODg4PDQ8ODw0PDQ8ODw0PDg4ODw0PDg8pDw4PKg8NDw0PDg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODw0PKg8qDw0PDg4ODw0PDg8NDw0PDg8NDw4OKg8qDw4ODg8NDw4PDQ8NDw4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg8pDw4PKg4ODw0PDg8pDyoPDg8pDyoPDg4ODw0PDg8pDw4PDQ8ODioPDg8NDw4ODg8NDw4PKQ8qDyoPDQ8qDyoPAA0F",
        "Comfort mode": "JgCKAg0NDw4PDQ8ODg4PAAM3cTgPKg8ODg4PDQ8qDw4ODg8NDw4PKQ8ODyoOKg8ODyoOKg8qDyoPKg4ODw0PKg8ODg4PDQ8ODw0PDg4ODw0PDg4ODw0PKg8ODg4PDQ8ODw0PDQ8ODw0PKg8qDykPKg8qDyoPKg4qDw4PDQ8ODioPKg8NDw4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg4ODw0PDg8NDw4ODg8NDyoPDg4qDyoPDQ8ODw0PDg4ODw0PDg8NDw4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8qDw4ODg8NDw4PDQ8NDw4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg8NDyoPDQ8qDw0PDg8NDw4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg4ODw0PDg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODykPDg8NDyoPKg8pDyoPKg8ABIxxOA8qDw0PDg8NDyoPDQ8ODw0PDg4qDw4OKg8qDw4OKg8qDyoPKg4qDw4PDQ8qDw0PDg8NDw4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODg4PDg4ODyoODg8NDyoPKg8qDg4PDg4ODw0PDg4qDyoPKg8NDw4ODg8ODg4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg8pDw4PKg4ODw0PDg8NDw4ODg8NDw4ODg8ODg4PDQ8ODg4PDQ8ODw0PKg8qDg4PDQ8ODw0PDg4ODw0PDg8NDw4OKg8qDw4ODg8NDw4ODg8NDyoPDg4ODw0PDg4ODw0PDg8NDw4ODg8qDw0PKg8NDw4ODg8qDyoODg8qDykPDg8NDw4ODg8qDw0PDg4ODyoPDQ8ODg4PDQ8ODg4PKg8qDioPKg8qDyoOAA0F",
        "Sensor mode": "JgCKAg4MDw4PDQ8ODw0PAAM2czgPKQ8ODw0PDg4qDw4PDQ8NDw4PKg8NDyoPKg4ODyoPKQ8qDyoPKg8NDw4PKQ8ODw0PDQ8ODw0PDg8NDw0PDg8NDw8OKQ8ODw0PDQ8ODw4ODw4pDw4PKRApDyoPKg8qDykPKhAqDg0QDQ8NDyoPKg8NDw0PDg8ODg8ODg4NDw4PDQ8PDQ8ODg4PDg4ODw0PDg0PDw4pDw4PKQ8ODw0PKg8NDw8ODQ8ODw4ODQ8ODw0PDw0PDg0PDw4ODg0PDw4ODg4PDg4qDw0PDw0PDg0PDw4ODg8NDw4NDw4PDg8NDg8ODg4NDw4PDg4PDikPDg8pEA4ODg4PDg4ODg4PDg8NDw4ODg4ODw4ODg4ODg8NDw8ODg4PDQ8ODg4PDg0PDg4PDg0PDw0PDg0PDw4ODg8NDw4NDw8ODg4PDQ8OKRANDw4ODw4ABIpzOA8qDw0PDQ8ODyoPDQ8NDw8ODQ8qDw0PKg8qDw0PKg8qDykQKQ8qDw0PDw4qDw0PDQ8ODw4ODg4PDg4ODw4NDw0PDg8NDw4PDQ8NDw8ODQ8PDSoPDg8NDyoPKg8qDw0PDQ8PDQ4PDQ8qDyoPKg8NDw0PDg8NDw0PDg8NDw8ODg4PDQ8ODg4PDg4ODg4qDw0PKg8ODw0PDQ8ODw4ODg8ODg4ODw4ODg8NDg8NDw4PDg4NDw8OKg8pDw4PDg4NDw4PDg4PDg0PDw0ODw0PKg8qDw0PDw0PDg4ODg8NDysODQ8PDg4ODQ8PDg4ODQ8QDQ4ODw4pDw4PKRANDw0PDg8pDyoPDg8pDyoPKg8ODg4PDQ8qDw0PDg8NDyoPDQ8ODw0PDQ8ODw0PKg8NDw8ODQ8NDw8OAA0F",
        "Powerful mode": "JgCKAg0ODw0PDg4ODw0PAAM3cjgPKg8NDw4ODg8qDg4PDQ8ODw0PKg8NDyoPKg8NDyoPKg8qDioPKg8NDw4PKg4ODw0PDg8NDw4ODg8NDw4PDQ8NDw4PKg8NDw0PDg8NDw4ODg8qDykPKg8qDyoPKg4qDyoPKg8qDg4PDQ8ODyoOKg8ODw0PDQ8ODw0PDg4ODw0PDg8NDw4ODg8NDw4ODg8NDw4PKg4ODw0PKg8qDw0PDg8NDw0PDg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODg4PDQ8ODw0PDg4qDw4PDQ8OFAgWBhANDw0PDQ8ODw0PDg4ODw0PDg8NDw4ODg8NDyoPDREoDw4ODg8NDw4PDQ8ODg4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg8NDw4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODw0PDg4ODyoODg8NDw4PDQ8qDyoPKQ8ABItzOA8pDw4PDQ8ODioPDg8NDw4ODg8qDw0PKg8qDw0PKg8qDioPKw4qDw0PDg4qDw4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg4ODw0PDg8NDw4ODg8NDyoPDg4ODyoOKg8qDw4ODg8NDw4PDQ8qDyoOKg8ODw0PDg4ODw0PDg4ODw4ODg8NDw4ODg8NDw4PDQ8qDw0PKg8ODg4PDQ8ODg4PDQ8ODw0PDg4ODw0PDg4ODw4ODg8NDw4OKg8qDw4ODg8NDw4PDQ8ODg4PDQ8ODg4PKg8qDg4PKg8NDw4ODg8NDw4ODg8NDw4PDQ8ODg4PDQ8ODg4PDQ8qDw4OKg8ODw0PDg4qDyoPDQ8ODw0PDg4ODw0PDg8qDg4PDQ8ODioPDg8NDw4ODg8qDyoOKg8ODykPDg8qDioPAA0F"
    }
}

I hope this information can be helpful in identifying the correct S21 commands or understanding why the D6 command is being rejected. Thank you for your continued support!

revk, 2025-06-11:
IR codes bear no relation to S21, sorry.