revk/xESP32-Faikout #748: Add Option to reboot on webpage shown when connected via wifi_ap

ConradWood, 2025-06-17:
**Daikin hardware**
S21 lead

**Is your feature request related to a problem? Please describe.**
After some maintenance work on the AC unit, the power was cycled. The faikin unit did also power cycle but did not reconnect to the wifi network. I was able to connect to the accesspoint the faikin module creates (Faikin-[mac]). On its webpage I was only able to change/update the wifi settings. Even after doing so, the module did not connect to wifi. I had to disconnect the power, wait for 30s and do this again. (This involves using the circuit breaker in a basement, restarting all other units too. not ideal)

**Describe the solution you'd like**
in cases like this it would be useful to reboot the faikin module from the webpage accesspoint menu.

**Describe alternatives you've considered**
Always connect to wifi. Figure out why it did not. This is difficult, because it happens very rarely and it only seems to happen to modules in the indoor units (not bench units). That makes it hard to gather logs.

**Additional context**
Had I reduced the "wifireset" parameter from default 3600 to 600 it probably would have reset itself whilst I was still in the basement fiddling with circuit breakers. Maybe that would help as well.

revk, 2025-06-17:
OK in next beta release in a few minutes.

ConradWood, 2025-06-17:
I also was surprised to find it did not connect. After (another) powercycle it *did* reconnect.
I even entered a different SSID to connect to. I thought maybe there's sth in the ESP NVRam preventing it from connecting to the SSID.
But that did not help either.
Only a powercycle.