revk/xESP32-Faikout #749: WebSocket multiple connections

Sigi-Bereuter, 2025-06-17:
I want to prevent using MQTT, so I tried to use the WebSocket directly as the Web Frontend does.

I connect to the WebSocket, that is hosted for the Web Frontend, which works. I am able to receive status messages and send commands. But it seems that from the point on when i connect with my own WS client the WebFrontened  ist not able to connect to the Websocket anymore - even worse the webserver at ipaddress:80 does not return the static html page at all,it does not anymore serve the GET requests.

I then need to stop my WS client and repower the faikin to get it respondign again.
So it seems the WebServer does not support multiple WS connections.

Has anyone experience with this ?

revk, 2025-06-18:
Indeed, not sure how many sessions we have it configured to support.

macmpi, 2025-07-04:
interesting in relation to https://github.com/revk/ESP32-Faikin/issues/299