revk/xESP32-Faikout #752: [FEATURE]

grbasta, 2025-06-20:
Good morning, I just wanted to know if it is necessary to flash since I purchased on Amazon, because after having connected it to my air conditioner on the CN_Wired port, I cannot find the faikin captive portal, by scanning the wi fi networks.
Thanks

grbasta, 2025-06-20:
no it doesn't turn on

revk, 2025-06-20:
Then double check wiring and voltages.

grbasta, 2025-06-20:
I will certainly do it

revk, 2025-06-20:
All boards are flashed and tested before they go to Amazon.

grbasta, 2025-06-20:
Hi sorry I tried to power it on the bench first at 5V respecting the pinout and no LED lights up and then at 12v but nothing happens, solutions?

grbasta, 2025-06-20:
Thank you very much at banco now it works and I also did the latest update via ota

closed 2025-06-21 by revk