revk/xESP32-Faikout #754: New user. System is offline (no data tx/rx) on an FTXS25D2VMW

PaskoMoto, 2025-06-20:
Hi.

I just connected the module to the S21 port of my interior unit, but the web interface says "system is offline".
I have read other issues similar to this and changed GPIO TX-RX pins to no avail. 
Reset to factory did not help either. 

Two weeks ago I tried for the first time, but did something weird (probably switching on/off the power too quickly) and had to bring the PCB to repair. Today I got it repaired and everything works as expected, except for the Faikin module. 

I have seen some S21 connections using only 4 pins, instead of the 5 I'm using (one-to-one as explained in the docs:
Would this be worth trying? 
And if it is:
Which one should I discconnect?

I must also say that this model was intalled back in 2005 and has a regular S21 port in the PCB. I'm using pinned cables. 

Thanks in advance.
Best regards.

P.S: I have two modules, but I'd rather not try the second one untill I try all options.

revk, 2025-06-20:
CN you check the voltages on all the pins?

revk, 2025-06-21:
If you can confirm your order details I can get another sent.

PaskoMoto, 2025-06-21:
I ordered via Amazon in Spain. Can it still be done?

revk, 2025-06-21:
I can ship you one, there is a risk of duty charge, but I'll list as a gift or replacement or something.

revk, 2025-06-21:
What was the amazon order number?

PaskoMoto, 2025-06-21:
171-9901369-2165900

revk, 2025-06-21:
That is not coming up. Do you have the full address, with town and any postal code?

PaskoMoto, 2025-06-21:
Yes.
But I'd rather keep it private. 😉

revk, 2025-06-21:
Understandable. Puzzled that I cannot find the order though. I think it is an Amazon re-sale via Spain, so I may not be able to see the order.

OK, so may be better return as faulty to Amazon and get replacement, sorry.

PaskoMoto, 2025-06-21:
Don't worry. I like the idea very much. I'll probably order more later.... 😉

PaskoMoto, 2025-07-09:
Hi.

I just want to add that I bought two new devices and now my three indoor units are connected without problems.

I will also close the issue.
Best Regards.

closed 2025-07-09 by PaskoMoto