revk/xESP32-Faikout #755: MQTT Setting dont apply (WIFI CONECTED, no MQTT

FigJam23, 2025-06-20:
Faikin Hardware

Self-built using an ESP32-D0WDQ6 development board (confirmed via esptool flash log).
Flashed using both esptool.py and ESPHome-Flasher. Also tested alternate .bin files from the Faikin GitHub release section and dev branches.

The device boots, enters AP mode, connects to Wi-Fi, and loads the web UI correctly on LAN (e.g., 192.168.1.38).
Daikin Hardware

Not yet connected to the Daikin unit.
Testing firmware and network integration on the bench first.
Planned connection: S21 to Daikin FTXS60KVMA.
Describe the Bug

After flashing, Wi-Fi setup and web UI work perfectly.
However, the MQTT configuration fails:

    Inputting valid IP, port (1883), username, and password into the web UI does not result in MQTT connection.

    No MQTT handshake or topic activity occurs (verified via MQTT Explorer and broker logs).

    Reboots or config changes result in NVS errors and occasional soft resets.

    Saving MQTT config appears to silently fail or not persist.

To Reproduce

    Flash Faikin ESP32 .bin using esptool.py or ESPHome Flasher

    Join AP, configure Wi-Fi → successful

    Access Web UI over LAN

    Configure MQTT (tried both IP and homeassistant.local)

    Click Save, device reboots

    No connection to MQTT broker, no discovery topics, no retained config

Expected Behavior

MQTT settings should persist and the device should connect to the broker (Mosquitto add-on in Home Assistant), publishing discovery/config topics.
Logs

From serial monitor:

E (69) RevK: NVS error ESP_ERR_NOT_FOUND
E (229) RevK: Not using LED GPIO 47
E (234) RevK: APSTA config mode start Faikin-7C9EBD483FB8
E (260) httpd: httpd_server_init: error in listen (112)
E (53447) RevK: Got IPv4 192.168.1.38 from Reaver
E (153082) RevK: Error ESP_ERR_NVS_INVALID_HANDLE
E (465099) wifi_init: Wi-Fi not stop

MQTT debug logs from broker show no connection attempts at all from the Faikin device.
Additional Context

    Board is confirmed to work (previously used with other projects)

    Broker IP 192.168.1.44 is active and accessible from other ESPs

    Tried both IP and homeassistant.local as broker hostname

    Erased flash between attempts using esptool.py erase_flash

    Tried alternate .bin files and boards (same outcome)

This seems to be a firmware-side NVS or MQTT config persistence bug on ESP32 builds.

[image]

FigJam23, 2025-06-20:
> The ESP_ERR_NVS_INVALID_HANDLE is likely a clue. Maybe try a full erase-flash and reflash it all?

Thanks for the tip! I’ve actually tried a few different things already:

Fully reflashed multiple times using the official .bin files from the Faikin repo.

Used different ESP32 boards (e.g. ESP-WROOM-32), all show the same MQTT issue.

I flashed using the ESP Flash Tool — maybe that’s part of the issue? I haven’t compiled my own .bin, just used prebuilt files.

After flashing, the device boots and Wi-Fi connects fine, but MQTT won’t connect.

I don’t always see the ESP_ERR_NVS_INVALID_HANDLE myself, but it may have shown up — I’ll try to confirm by capturing serial logs directly.

Do you recommend a better flash method or compiling from source? I suspect either:

1. MQTT credentials are being ignored or not saved properly, or

2. The flash method isn’t writing everything correctly, especially NVS.

I’m happy to test with esptool.py if that’s more reliable. Let me know your thoughts!

FigJam23, 2025-06-21:
Am I doing something wrong?
```
start detect chip...please wait
chip sync ...
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: Wi-Fi, BT, Dual Core + LP Core, 240MHz, Vref calibration in eFuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c9ebd483fb8
Manufacturer: d8
Device: 4016
Status value: 0x200202
Detected flash size: 4MB
```
```

start read flash...please wait
read flash address: 0x000
read flash size: 0x10000
chip sync ...
MAC: 7c9ebd483fb8
Manufacturer: d8
Device: 4016
Status value: 0x200202
Detected flash size: 4MB
chipInfoPanel.read_flash_start_func.<locals>.flash_progress() takes 2 positional arguments but 3 were given

```
[image]

revk, 2025-06-21:
You MQTT server is not expecting mqtts only is it?

By default this is mqtt

(You can add certs for mqtts in advanced settings)

FigJam23, 2025-06-21:
/

> You MQTT server is not expecting mqtts only is it?
> 
> By default this is mqtt
> 
> (You can add certs for mqtts in advanced settings)

Thanks for the quick reply!

No, my MQTT server is not expecting mqtts. I'm using the standard mqtt protocol without any certificates. Here's my current setup (Hass.io Mosquitto add-on):

logins:
  - username: mqtt
    password: mypw
require_certificate: false
certfile: fullchain.pem
keyfile: privkey.pem
customize:
  active: false
  folder: mosquitto
allow_anonymous: true

The issue I'm seeing is that when I try to enter MQTT settings via the web UI, the device often reboots and loses Wi-Fi settings. I then need to reconnect to the AP and start over, which is frustrating.

Also, I'm unsure if my flashing process is the problem:

If I flash only the main .bin file using the ESP-Flasher, the AP works correctly, and I can access the web UI after connecting it to Wi-Fi.

But if I use the ESP Download Tool and flash the combined files (bootloader, partition table, and main binary), no AP shows up at all, and I can’t configure anything.

Is there a correct flashing method or file structure for the ESP8266 version? Could this be why MQTT setup isn't saving properly?

Thanks!

revk, 2025-06-21:
Well I have the full idf and build and use the flash command, but you would need all the bin files as one is partition table.

FigJam23, 2025-06-21:
> Well I have the full idf and build and use the flash command, but you would need all the bin files as one is partition table.

Thanks so much for your help and info – really appreciated. I did try again with the full IDF and build process, but to be honest it’s a bit beyond me at this stage 😅. I’m not sure if trying on Windows is half the problem, but even after cloning the repo and following the steps, I just keep hitting errors and end up wiping it all and starting over.

I love the idea of this project and how cleanly it integrates with Home Assistant – it’s exactly what I was looking for! But I think I’ll explore some other options for now like a Daikin WiFi module or maybe go down the IR blaster path. I had this vision of just flashing a .bin and voilà – done – but turns out it’s a bit more involved than I expected 😄.

Thanks again for taking the time to help!

FigJam23, 2025-06-21:
I have order a few of these boards, might have another play when they arrive 😀 

[image]

FigJam23, 2025-06-21:
> Maybe order some of the Faikin modules, see tindie.revk.uk

I'm still considering that option, unfortunately I'm in New zealand. the usd to nzd and shipping adds up. Our dollar dosnt go far here 😢. I also like to be able to do things myself if I can. I haven't given up yet, still exploring options. Realy appreciate this project and your assistance. Very much appreciated.

FigJam23, 2025-06-21:
[image]

revk, 2025-06-22:
Yeh, you use make and that builds the settings.

If that works the release binaries should also work.

If using your own build and signing key you won't be able to upgrade from ota.revk.uk

FigJam23, 2025-06-22:
> Yeh, you use make and that builds the settings.
> 
> If that works the release binaries should also work.
> 
> If using your own build and signing key you won't be able to upgrade from ota.revk.uk

Cool thank you, it's now the  first night testing and autimations set in hassio and wow what a difference having this level of control. Sorry I didn't buy the offisal devic.  I' would be  happy to send a donation in support of this project, I dont have alot but every bit helps.  Have you a link somewhere for that type of thing.

revk, 2025-06-22:
Don't worry. Glad it is working.

closed 2025-06-22 by revk