revk/xESP32-Faikout #756: MQTT command for Faikin Auto Mode

christianmue1980a, 2025-06-22:
Hi,

thanks a lot for all the work that you put into this project! I'm very happy that it exists and works.

I want to set these three parameters via MQTT, but cannot find the correct topics / naming in the docs. Can you help me?

[image]

And one little question: When using the Faikin Auto Mode, I notice that the Fan mode sometimes jumps between Night (my preferred one) and Auto back and forth. Any ideas where this comes from?

christianmue1980a, 2025-06-22:
This does not work:

Topic: `Faikin/AC123/command/control`
(prefixapp and prefix host enabled)

```
{
  "target": 23,
  "marin": 1
}
```

revk, 2025-06-23:
There are two types of control.

- The settings, as per your image (they are settings, so use `setting/...` and the setting field itself).
- Remote control - this works by sending that `control` message regularly (e.g. every minute) - it hides those web settings even.
- Also, now, can be remote via BLE from Faikin Remote.

revk, 2025-07-01:
I'll try and explain.

For a start, the structure of the topic is configurable, but I'll assume you have not changed that.

The topic normally has there parts:-

- The start is either `command` or `setting`. You do not want `setting` as that changes settings in the Faikin, which can be anything from wifi settings to GPIOs. Settings are usually best changed in the web interface. You want `command` which tells the Faikin to do something. `power` is not a Faikin setting.
- The name, which I assume in your case is `daikinac`
- In the case of a command, there is also the command you want, which in this case is `control`

So you want `command/daikinac/control` and a JSON payload. That payload can contain controls for the AC. `power` is a control. Do a JSON payload of `{"power":true}` would work. Note, as JSON, it is `"power"` not `'power'` and `true` not `'true'` or `"true"`.

christianmue1980a, 2025-07-01:
Hi @revk ,
thanks a lot for the clarification and sorry for my delayed response. Worked fine for me so far.

Do you have a recommendation how I can achieve the following behavior?
* In most rooms, I want the fan to be as silent as possible (i.e. night mode). 
* I would prefer to have the fan running constantly (more or less, as needed to reach the target) in night mode rather than hearing the fan going on and off every few minutes (with more rpm as in Auto mode)

I tried it with the following settings with Faikin auto mode:
* `nofanauto = true`
* selected `fan = night` 
* `target = 22` and `range = +/- 0.5`
* `env = 23`
* ...so the target is not reached, however, the fan goes on and off in night mode as well instead of continue cooling down to 22.5

Any ideas?

And what I also noticed: I'm sending the `env` temperature every second to a MQTT topic (with `retain = true`) and using the faikin feature to subscribe to this topic. However, a few times per minute, the UI shows `Env ---`, i.e. no value for a few seconds. Does this have a direct influence to the behavior mentioned above?

revk, 2025-07-01:
Retain won't help, it looks for messages being sent regularly. Every second is overkill.

I'd send target and margin and env all in one message, maybe every minute.

christianmue1980a, 2025-07-01:
Ok, so this one every minute to `Faikin/acname/command/control <v>`

```
{ "power": true, "mode": "C", "fan" : "Q", "swingh": false, "comfort": true }
```

But the target for Faikin auto is a setting, right?

Currently, I'm using this topic:
`Faikin/acname/setting/autot <v>` and set the value e.g. to `22`
and 
`Faikin/acname/setting/autor <v>` and set the value e.g. to `0.5`
which is then reflected in the UI

[image]

Correct?