revk/xESP32-Faikout #759: [BUG] FVXM Heating floor flap move in cooling mode

Ulf349, 2025-06-26:
**Faikin hardware**
Self created Faikin from ESP S3

**Daikin hardware**
Daikin FVXM25, connected via S21 lead

**Describe the bug**
Faikin is in cooling mode with fan night and target 23°C with external BLE temperature Humidity sensor with Faikin auto-mode range +/-0,5°C
If the room temperature increases, the floor heating flap opens, the Daikin cool down the room with Set temperaure 18°C, the top flap permanentis open and not moving. 
If the BLE room temperature is decreased to target range, the Set temperature change to 28°C and the floor heating flap close. Sometimes the floor heating flap open and close in 2 minutes 3 times.

I can´t see the position in the Onecta app, because normally the floor heating flap can only be moved in heating mode.

**To Reproduce**
Steps to reproduce the behavior:
1. increase or decrease the target temperature, that the Daikin starts or ends the cooling, the bottom heating floor flap moves.

**Expected behavior**
I don´t know, why that happens. In the normal case the bottom heating floor flap can only moves in heating mode. I think in the Faikin software is a bug.
I have 9 Faikins, 6x in FTXM with 1 flap-motor and 2 swing possibilities which can be adjusted via the Faikin module, 3x FVXM with 2 flap-motors (one for the top flap and one for the bottom flap). Maybe there is the software problem because both models have some motors.

**Additional context**
I didn´t know, that the Faikin module can move the bottom floor flap.

revk, 2025-06-26:
I don't really know about floor flaps much. You can set `dump` to log the S21 messages used to MQTT. Maybe you can catch the message being sent?

revk, 2025-06-26:
That looks like you are only logging the `status` message - there should be more going to MQTT than that.

revk, 2025-06-26:
Does `swing` (h or v) control the flap, or is the flap something else that is not related (if so, very puzzled as we don't have a *flap* control.

Ulf349, 2025-06-26:
In the Daikin FVXM are 2 flaps available. The top flap and a bottom flap. The bottom flap can original works only in heating mode.

swing v moves the top flap - that works

swing h have no function - so I disabled the swing h

revk, 2025-06-26:
May be worth trying each of `1`, `2`, `4`, `8` in all 3 places where a `0` in `D50?00` see if any work the bottom flap.

Ulf349, 2025-06-26:
These are done, but no moving the bottom flap.

D50?00
D51?00
D52?00
D54?00
D58?00

D50?00
D50?01
D50?02
D50?04
D50?08

D50?00
D50?10
D50?20
D50?40
D50?80

Is it also necessary to mix all the number like this: D52?48

revk, 2025-06-26:
OK sounds like the bottom flap is not on that command.

Maybe try `D5????` as a last resort.

Otherwise we have to find, somehow, where the bottom flap is, in what message.

sam-m7, 2025-07-10:
Hi, we also have FVXM. If I understand the problem correctly, I don't think this has anything to do with the Faikin commands.

When you just set the target temperature of the AC and leave it (also when just using the remote) and the difference between the target and the current room temperature (measured by the ACs internal sensor) is huge, the lower flap will be opened. When the temperature difference gets smaller it at some point closes the lower flap, so cold air only comes out from the top of the AC. When you just set a temperature and don't change it, the lower flap will usually only open and close once. I guess this is just a function of the ACs internal logic to have more airflow (and therefore possibly greater efficiency) when the temperature difference is high. And only use the upper outlet when the temperature difference is small and less cooling is needed to minimize the temperature difference in the room between the ceiling and floor and let convection do it's thing. I don't think this inherently comes from the S21 commands, but rather from changing temperature set points in Faikin auto mode frequently and to a great extend and the ACs logic not expecting this.

revk, 2025-07-11:
There are settings for how far off the Faikin auto mode sets the temperature, so you may be able to fine tune this.