revk/xESP32-Faikout #772: [FEATURE] Zigbee Variant?

Ansuel, 2025-07-12:
The question is simple... Is it possible to have a zigbee variant? I honestly hate polluting wifi network with iot devices and it would be ideal to have everything on zigbee.

Do you think it's possible? Considering it's the same exact frequency and i think esp32 can run zigbee...

I'm also open to fund the variant just to give you more reason for it 👍

revk, 2025-07-12:
The ESP32-S3 I use don't do zigbee. Some other models do.

I do have working via BLE from a remote control if that helps.

Ansuel, 2025-07-12:
I wonder what are the effort of using esp32-h2 or c2/c3 instead of s3. Are s3 feature needed for faikin? (for example dual core, high cpu freq, lots of gpio...)

Just asking if you have any additional tech info of any specific requirement of s3. I might be crazy enough to adapt the codebase to use different chip

revk, 2025-07-13:
I am not sure those have WiFi, do they? The current design is very much WiFi, http web control, and MQTT.

I have no idea on the internals of Zigbee, sorry, so no idea how much work it is.

FloppyDance, 2025-07-27:
@revk What about connecting it through Ethernet or Modbus?

revk, 2025-07-28:
It would need an Ethernet chip. Ethernet support is good in the ESP IDF and could do MQTT. I'd have to find a suitable board to test.

Modbus is possible but we are back to what protocol? It is not MQTT is it.

FloppyDance, 2025-07-29:
Adding Ethernet port would be a very good step forward.

Regarding bus support, you can think about a master-slave solution. Each air conditioner has its own slave board which is connected through a 4-wire bus to the master board which is connected via Ethernet or Wifi to the router. This should improve connection where Wifi is poor or not available.

revk, 2025-07-29:
I suspect you are missing somewhat how this all works, sorry.

techman83, 2025-07-31:
> However, please consider the idea of implementing at least an Ethernet port on Faikin

There are some off the shelf esp32 ethernet devices. In theory you could build a bespoke device to fit your needs and flash the faikin firmware. Wifi is just the physical layer, MQTT/HTTP etc don't really care about that layer being Wifi or Ethernet based.