revk/xESP32-Faikout #773: [BUG]Intelligent Eye and LED controls don't work at FTXM20A5(Perfera)

epiniguin, 2025-07-14:
**Faikin hardware**
Faikin-C3: 81f0280 2025-06-21T07:29:41 S21

**Daikin hardware**
Perfera Multi-split
Indoor units FTXM20A5
Outdoor unit 2MXM40A2

**Describe the bug**
The "Sensor mode" and "LED" controls don't work as they should work.
I mean that "sensor mode" should turn on/off intelligent eye function, and "LED" should turn off/high/low LED.
Instead of this behavior these two switches react on LED state as below:

LED state | Faikin LED switch | Faikin Sensor switch
-----------|--------------------|----------------------
Off | Off | On
Low | On | On
High | On | Off

Besides, the AC doesn't react to Faikin LED switch toggling. After Faikin LED switch toggled it comes back to the previous state in some seconds, and the AC LED doesn't change its state.

**To Reproduce**
1. Change LED and Intelligent EYE via remote control or Onecta.
2. See changing in Faikin "Sensor" and "LED" controls.
3. Toggle Faikin LED switch.
4. Wait 10 seconds, and see to the Faikin LED switch state.

**Expected behavior**
Sensor switch should toggle Intelligent Eye function, LED switch should toggle AC LED.

**Additional context**

revk, 2025-07-14:
I think we need debug of the S21 messages involved, then I can try and find what is needed.

There is a `debug` and a `dump` setting and logging to MQTT which may help.

epiniguin, 2025-07-14:
comfort off
eye on
led off

```{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"0\u00B00<","G7":"00P5","GA":"D120","GB":"0331","GG":"0500","GK":"q{51","GM":"F830","GN":"0000","GP":"1600","GQ":"2600","GS":"4000","GT":"2000","SG":"A","Sa":"033+"}```

comfort on
eye on
led off

```{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"p\u00B00<","G7":"00P6","GA":"D120","GB":"0331","GG":"0600","GK":"q{51","GM":"0930","GN":"0000","GP":"1600","GQ":"2600","GS":"4000","GT":"2000","SG":"A","Sa":"533+"}```

comfort off
eye off
led off

```{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"0\u00B00<","G7":"00P5","GA":"D120","GB":"0331","GG":"0700","GK":"q{51","GM":"0930","GN":"0000","GP":"1600","GQ":"2600","GS":"4000","GT":"2000","SG":"A","SX":"062+"}```

comfort on
eye off
led off

```{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"p\u00B00<","G7":"00P6","GA":"D120","GB":"0331","GG":"0800","GK":"q{51","GM":"1930","GN":"0000","GP":"1600","GQ":"2600","GS":"4000","GT":"2000","SG":"A","SD":"000"}```

comfort on
eye on
led high

```{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"p\u00B004","G7":"00P6","GA":"D120","GB":"0331","GG":"0B00","GK":"q{51","GM":"2930","GN":"0000","GP":"1600","GQ":"2600","GS":"4000","GT":"2000","SG":"A","SI":"521+"}```

comfort on
eye on
led low

```{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"p\u00B008","G7":"00P6","GA":"D120","GB":"0331","GG":"0C00","GK":"q{51","GM":"2930","GN":"0000","GP":"1600","GQ":"2600","GS":"4000","GT":"2000","SG":"A","SL":"650"}```

epiniguin, 2025-07-15:
- Yes, I used the remote for changing these settings.
- `noled` and `nosensor` are disabled.
- I can control `comfort` from Faikin.
- I can't control `eye` from Faikin.
- I can control `eye` from Onecta.
- If I toggle the `sensor` switch in Faikin, the AC LED state is changed, and after 2-3 seconds the Faikin LED switch is automatically switched to opposite state: if I turn on `sensor`, `LED` switch is turned off, if I turn off `sensor`, `LED` switch is turned on.

I guess that it should be possible to control `eye` from Faikin, because Onecta can control it, and Onecta works also via S21.
Maybe you don't debug all G sensors to MQTT, and there is something else.

revk, 2025-07-15:
I think `eye` is `sensor`, but if, as you say, `sensor` is changing LED we definitely have something odd. They are in the same message.

But I am not seeing `eye` change anything in the messages you posted, so could be a message we are not using (yet).

@Sonic-Amiga - do you have any ideas?

epiniguin, 2025-07-15:
> But I am not seeing `eye` change anything in the messages you posted, so could be a message we are not using (yet).

I also think that there is something else.

Nevertheless the `LED` control should have 3 state: off, low, high.

revk, 2025-07-15:
The current code for LED is on/off, but depends on `dark` setting as well, and combined it should do three states. In theory.

epiniguin, 2025-07-15:
As I said, currently `LED` control doesn't switch AC LED in Perfera at all. After toggling `LED` switch comes back in some seconds.
`Dark` switch was ON in my tests.

revk, 2025-07-15:
Yes, is it any different if you try `nosensor`.

But the underlying issue is the same, we need to work out the correct S21 messages, @Sonic-Amiga is probably best to confirm. Then I can make more changes.

epiniguin, 2025-07-15:
> Yes, is it any different if you try `nosensor`.

If I turn on `nosensor`, `LED` switch combined with `Dark` works as it should.

epiniguin, 2025-07-15:
> What kind of wi-fi module do you use for Onecta ? I am going to Spain for vacation this August, so may try my luck in an A/C shop...

Perfera has a built-in module.
Onecta APP tells it's BRP069C4x.
Onecta firmware version is 1.39.0

Sonic-Amiga, 2025-07-29:
I have BRP069C41, but i was unable to get it working with S21 protocol. It does recognize S21, but only goes online in Onecta with X50 protocol. I use simulators, i also tried to connect it to my real A/C unit, the result is the same.
@epiniguin May i see a photo of built-in module ?

epiniguin, 2025-08-01:
@Sonic-Amiga I'm sorry for a delay.
I need to remove an indoor unit cover to make a photo of the built-in module.
I'm going to do it today.

cracyfloyd, 2025-09-27:
Same here. The eye and LED on/off do not work with Faikin. It works with the remote control and Conecta app.

Faikin hardware
Faikin-S3-mini-n8 S21 ota channel: ota.revk.uk 

Daikin hardware
Perfera Multi-split
Indoor units FTXM35A5V1B