revk/xESP32-Faikout #777: Hardware from Amazon damaged - improper packing

oliveres, 2025-07-24:
Hello, I just received envelope from Amazon and ESP32 chip is damaged because PCB is packed in envelope, without any protection. Strongly advised to change packing requirements.

[image]
[image]

revk, 2025-07-24:
Amazon should do a return no problem, but we have shipped a lot of these with no problem - this is very unusual, but thanks for letting me know.

Just to clarify, was this not also in the thick card envelope Amazon usually use?

closed 2025-07-24 by revk

oliveres, 2025-07-24:
Thin paper envelope, without bubble wrap.

This packaging is really inappropriate in my opinion. 

I'll order it from Tindie.

[image]

revk, 2025-07-24:
As I say, this is the first instance of Amazon breaking one - useful to know though, thank you.

FYI, for Tindie, I ship direct and that is in bubble wrap not card.