revk/xESP32-Faikout #778: [BUG] System is offline (no data tx/rx) on my FTXJ20A2V1BW, and no loopback detected

b-abadie, 2025-07-24:
Hello,

**Faikin hardware**

I Have a Faikin I bought from Amazon France in December 2024
On it, it says 2024-05-16 and what seems to be a serial number : 5060634238151

**Daikin hardware**

My aircon is a Daikin FTXJ20A2V1BW, and the Faikin module is plugged on the previously-free S21 port.

**Describe the bug**

On my model, there appears to be no 12V on PIN4 so I plugged aircon's PIN1 (5V) on Faikin's PIN4. It can boot fine and I was able to get it to join my SSID and connect to my MQTT broker.

However, the webUI says "System is offline (no data tx/rx)".

I tried to bridge rx and tx on the Faikin, and it does not say anything about being in loopback test. Is it still supposed to ?

I tried plugging an oscilloscope on the Faikin's PIN2 and PIN3. While I see a signal on PIN3, PIN2 always stays at 0V (which is weird because when there is no Faikin, it stays at 5V).

I cannot even guess which of the Faikin or the aircon is faulty. No loopback detected would point to the Faikin, but the absence of signal on PIN2 would point to the aircon ? Or maybe my pinout is wrong (do I need Faikin's PIN1 plugged somewhere ?)

I would greatly appreciate any help. Thanks !

**Additional context**

I had a little bit of trouble finding the correct pinout (I did not know about the little triangle indicating PIN1 on the aircon S21 port...). I hope I did not damage anything.

**Photos**

Faikin model : 

[image]

Aircon S21 port (with an EKRS21 cable) : 

[image]

Faikin in loopback mode : +5V on PIN4, GND on PIN5, and PIN2/PIN3 bridged : 

[image]

Oscilloscope probes : red hook is the green line on the scope, and green hook is the blue line (yeah, I know...) :

[image]

Scope screen when in loopback mode (both lines at 0V) : 

[image]

Faikin back in a classic connection (with rx/tx linked to the aircon) : 

[image]

Scope screen when connected to the aircon (signal on PIN3, nothing on PIN2) : 

[image]

revk, 2025-07-25:
Sorry for the delay.

The tx/rx is driven open drain with a pull up. We did find some Diakin units don't have a pull up on their tx side so we need a pull up in the Faikin (pin 2). I added a pull up and diode to the design some time ago - but I think your board pre-dates that. This also explains why loopback does not work.

When did you buy it, by the way? I wonder if Amazon have some old stock still.

Anyway, there is a relatively simple fix - a resistor from the 5V supply to pin 2 - something like 10k should be fine. Do you have one?

If you cannot get working, let me know your address (email pct at aa.net.uk) and I'll send a later board.

closed 2025-07-25 by revk

revk, 2025-07-25:
Thanks for confirming, that is a shame, but not surprising. I do also sell the cables on their own on tindie if that is anything to help. I hope to have on Amazon soon, but they are taking weeks to fix an issue with the listing.

b-abadie, 2025-07-25:
Unfortunately I already bought a few EKRS21 cables (at a price I would rather not discuss 😢), I did not know about yours... Next time I guess !

b-abadie, 2025-07-25:
Well, I soldered a 10k resistor between pin2 and pin4. I think I managed a decent job given my lack of soldering skill, and I measure 9.78 kohm between pin 2 and pin 4 on the white connector prongs (so at least the resistor-connector link is OK).

I see no change however : no loopback detected, and when connected to the aircon, I see no tx signal...

... well, actually I managed to catch what looks like a signal but the levels do not cross 400mV. Is it some kind of interference from the Rx measure, or does it confirm your theory of the absence of pull-up resistor (and I did a terrible soldering job) ?

The resistor : 

[image]

The mysterious signal : 

[image]

revk, 2025-07-25:
Email me your address

revk, 2025-07-26:
nice export, so data both ways, if Faikin does not work it is perhaps faulty. Email me and I'll send a new one. The latest designs actually have a final connector loop back test that is done on every one before going out. Always improving the design.

b-abadie, 2025-07-28:
Hello,
I sent you an email friday evening, did you get it ?

b-abadie, 2025-07-28:
Oh, you wrote "pct" instead of "pcb" on your original message (and I did not get a bounce so I guess "pct" is assigned). I will send again.