revk/xESP32-Faikout #779: [BUG] Temperatures are not updating till rebooting (sometimes twice)

borgqueenx, 2025-07-25:
**Faikin hardware**
See photo's.
[image]
[image]

Faikin-S3-MINI-N8: 81f0280

**Daikin hardware**
DAIKIN ATXC25D/ARXC25D

S21 lead.

**Describe the bug**
Temperature is not updating in the web UI and on MQTT. i can send commands to control the Daikin fine.  But temperatures are updating in web UI and in the web interface only when rebooting. and often, after a reboot, the faikin shows no temperature data. i need to reboot again, then it shows the updates outdoor and indoor unit values. If i do not reboot, it will not update any value.

**To Reproduce**
Simply setup faikin for local wifi and setting up mqtt. works in home assistant instantly.

**Expected behavior**
I expect temperature values to be polled and updated.

revk, 2025-07-31:
There are options to output debug and dump communications to MQTT, that may show the S21 messages or related, and so maybe give some clues.

revk, 2025-08-01:
That logging is just some of the polling, so needs more.

If you change temp on the AC by other means, does the Faikin show the new temp?

We are looking for logs with `D1`, `F1` and `G1`

revk, 2025-08-01:
`info/faikinbedroom/tx` and `info/faikinbedroom/rx` and `error/faikinbedroom/comms` should cover it.

borgqueenx, 2025-08-03:
ok, i tried to do it with a MQTT explorer, copying pasting everything from history, hope it can be useful.

TX:
`3-08-2025 13:05:37(-2.27 seconds) 
{"protocol":"S21","dump":"0246418703","FA":""}
03-08-2025 13:05:37(-0.1 seconds) 
{"protocol":"S21","dump":"0246377D03","F7":""}
03-08-2025 13:05:37(-0.11 seconds) 
{"protocol":"S21","dump":"0246367C03","F6":""}
03-08-2025 13:05:37(-0.1 seconds) 
{"protocol":"S21","dump":"0246357B03","F5":""}
03-08-2025 13:05:37(-0.1 seconds) 
{"protocol":"S21","dump":"0246347A03","F4":""}
03-08-2025 13:05:37(-0.1 seconds) 
{"protocol":"S21","dump":"0246337903","F3":""}
03-08-2025 13:05:36(-0.11 seconds) 
{"protocol":"S21","dump":"0246327803","F2":""}
03-08-2025 13:05:36(-0.11 seconds) 
{"protocol":"S21","dump":"0246317703","F1":""}
03-08-2025 13:05:34(-2.27 seconds) 
{"protocol":"S21","dump":"0246418703","FA":""}
03-08-2025 13:05:34(-0.1 seconds) 
{"protocol":"S21","dump":"0246377D03","F7":""}
03-08-2025 13:05:34(-0.11 seconds) 
{"protocol":"S21","dump":"0246367C03","F6":""}
03-08-2025 13:05:34(-0.09 seconds) 
{"protocol":"S21","dump":"0246357B03","F5":""}
03-08-2025 13:05:34(-0.11 seconds) 
{"protocol":"S21","dump":"0246347A03","F4":""}
03-08-2025 13:05:34(-0.1 seconds) 
{"protocol":"S21","dump":"0246337903","F3":""}
03-08-2025 13:05:33(-0.11 seconds) 
{"protocol":"S21","dump":"0246327803","F2":""}
03-08-2025 13:05:33(-0.11 seconds) 
{"protocol":"S21","dump":"0246317703","F1":""}
03-08-2025 13:05:31(-2.27 seconds) 
{"protocol":"S21","dump":"0246418703","FA":""}
03-08-2025 13:05:31(-0.1 seconds) 
{"protocol":"S21","dump":"0246377D03","F7":""}
03-08-2025 13:05:31(-0.11 seconds) 
{"protocol":"S21","dump":"0246367C03","F6":""}
03-08-2025 13:05:31(-0.1 seconds) 
{"protocol":"S21","dump":"0246357B03","F5":""}
03-08-2025 13:05:31(-0.1 seconds) 
{"protocol":"S21","dump":"0246347A03","F4":""}
03-08-2025 13:05:31(-0.1 seconds) 
{"protocol":"S21","dump":"0246337903","F3":""}
03-08-2025 13:05:30(-0.11 seconds) 
{"protocol":"S21","dump":"0246327803","F2":""}
03-08-2025 13:05:30(-0.11 seconds) 
{"protocol":"S21","dump":"0246317703","F1":""}
03-08-2025 13:05:28(-2.27 seconds) 
{"protocol":"S21","dump":"0246418703","FA":""}
03-08-2025 13:05:28(-0.11 seconds) 
{"protocol":"S21","dump":"0246377D03","F7":""}
03-08-2025 13:05:28(-0.1 seconds) 
{"protocol":"S21","dump":"0246367C03","F6":""}
03-08-2025 13:05:28(-0.09 seconds) 
{"protocol":"S21","dump":"0246357B03","F5":""}
03-08-2025 13:05:28(-0.12 seconds) 
{"protocol":"S21","dump":"0246347A03","F4":""}
03-08-2025 13:05:28(-0.1 seconds) 
{"protocol":"S21","dump":"0246337903","F3":""}
03-08-2025 13:05:27(-0.11 seconds) 
{"protocol":"S21","dump":"0246327803","F2":""}
03-08-2025 13:05:27(-0.11 seconds) 
{"protocol":"S21","dump":"0246317703","F1":""}
03-08-2025 13:05:25(-2.26 seconds) 
{"protocol":"S21","dump":"0246418703","FA":""}
03-08-2025 13:05:25(-0.11 seconds) 
[... 425 lines cut ...]
03-08-2025 13:04:14(-2.49 seconds) 
{"protocol":"S21","timeout":true}
03-08-2025 13:04:13(-0.5 seconds) 
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
03-08-2025 13:04:11(-2.52 seconds) 
{"protocol":"S21","timeout":true}
03-08-2025 13:04:10(-0.5 seconds) 
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
03-08-2025 13:04:08(-2.51 seconds) 
{"protocol":"S21","timeout":true}
03-08-2025 13:04:07(-0.5 seconds) 
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
03-08-2025 13:04:05(-2.47 seconds) 
{"protocol":"S21","timeout":true}
03-08-2025 13:04:04(-0.51 seconds) 
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
03-08-2025 13:04:02(-2.53 seconds) 
{"protocol":"S21","timeout":true}
03-08-2025 13:04:01(-0.5 seconds) 
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
03-08-2025 13:03:59(-2.51 seconds) 
{"protocol":"S21","timeout":true}
03-08-2025 13:03:58(-0.51 seconds) 
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
`

revk, 2025-08-03:
It looks like `FA` gets a `G[null]` response which is not a response I have ever seen. That may be a red herring though, as we only send `FA` in debug anyway.

All the `G1` are `03F7` which looks like mode *cool* and temp 21℃

Did you change temp during these dumps at all? Was it 21℃ when doing the dumps.

borgqueenx, 2025-08-03:
iit started at 21 degrees, and i changed the temperature up and down. however i am uncertain but fear the logs were not all kept in the MQTT explorer. i need a way to save them to a txt file over a specific period of time or until i simply stop logging. maybe @Jxhannes can gather some logs?

revk, 2025-08-03:
It would help to get different `G1` responses logged, I can then look and see if there is something wrong somehow.

Jxhannes, 2025-08-03:
I would like to help but I won't be home until 15th of August. Until then I can't help

Jxhannes, 2025-08-15:
As promised I am back at home and got some log data. The tiomes are different because I ahd to do it topic by topic, but in every case I started logging, went to the AC and tunred it on/opff and switched the temperature and sometimes the fan settign around. Hope this helps.

Currently I am not even able to control my AC by the Faikin Modul. If I turn it on by the remote teh faikin shows it correctly in HA but I cannot control anything from HA. Before I went on holidays I could control it over Faikin, so no idea why not anymore. 

comms
15.08.2025 09:41:14(-2.5 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:41:13(-0.51 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:41:11(-2.49 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:41:10(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:41:08(-2.52 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:41:07(-0.49 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:41:05(-2.47 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:41:04(-0.52 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:41:02(-2.51 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:41:01(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:59(-2.49 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:58(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:56(-2.5 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:55(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:53(-2.49 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:52(-0.51 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:50(-2.48 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:49(-0.53 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:47(-2.51 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:46(-0.49 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:44(-2.5 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:43(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:41(-2.5 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:40(-0.46 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:38(-2.54 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:37(-0.51 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:35(-2.4 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:34(-0.58 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:32(-2.5 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:31(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:29(-2.5 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 09:40:28(-0.51 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 09:40:26(-2.5 seconds)
[... 480 lines cut ...]
{"protocol":"S21","dump":"024731303348426503","G1":"03HB"}
15.08.2025 09:44:46(-2.27 seconds)
{"protocol":"S21","dump":"024700000000004703","G":"\u0000\u0000\u0000\u0000"}
15.08.2025 09:44:46(-0.1 seconds)
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
15.08.2025 09:44:46(-0.11 seconds)
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
15.08.2025 09:44:46(-0.11 seconds)
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
15.08.2025 09:44:46(-0.1 seconds)
{"protocol":"S21","dump":"024734302080307B03","G4":"0 \u00800"}
15.08.2025 09:44:46(-0.12 seconds)
{"protocol":"S21","dump":"024733303030303A03","G3":"0000"}
15.08.2025 09:44:46(-0.1 seconds)
{"protocol":"S21","dump":"024732343A8080E703","G2":"4:\u0080\u0080"}
15.08.2025 09:44:45(-0.11 seconds)
{"protocol":"S21","dump":"024731303348426503","G1":"03HB"}
15.08.2025 09:44:43(-2.26 seconds)
{"protocol":"S21","dump":"024700000000004703","G":"\u0000\u0000\u0000\u0000"}
15.08.2025 09:44:43(-0.1 seconds)
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
15.08.2025 09:44:43(-0.11 seconds)
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
15.08.2025 09:44:43(-0.11 seconds)
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}

borgqueenx, 2025-08-15:
I could also control the daikin fine until some days ago where i noticed i cant anymore. The status of the AC (except temperatures) are send to faikin fine, but i can no longer control the daikin with home assistant/mqtt.
In the faikin web dashboard i can switch buttons but they are also not working.

Jxhannes, 2025-08-15:
[image]

I just discovered what the problem is. When debug is activated, you no longer can control anything and just get Feedback .. 

I turned it off and now it's working again fine, except the current temperature reading

revk, 2025-08-15:
That should not be the case!

revk, 2025-08-15:
Dod you manage to get a dump of `F1`/`D1`/`G1` messages when actually changing the settings on the AC?

Jxhannes, 2025-08-15:
here you go, most of the messages are while changing the temperature via the remote

rx

Nachricht 88 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 87 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024731313346426403","G1":"13FB"}
QoS: 0 - Retain: false
Nachricht 86 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"02474D303030305403","GM":"0000"}
QoS: 0 - Retain: false
Nachricht 85 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
QoS: 0 - Retain: false
Nachricht 84 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
QoS: 0 - Retain: false
Nachricht 83 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 82 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024731313340425E03","G1":"13@B"}
QoS: 0 - Retain: false
Nachricht 81 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"02474D303030305403","GM":"0000"}
QoS: 0 - Retain: false
Nachricht 80 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
QoS: 0 - Retain: false
Nachricht 79 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
QoS: 0 - Retain: false
Nachricht 78 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 77 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024731313340425E03","G1":"13@B"}
QoS: 0 - Retain: false
Nachricht 76 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"02474D303030305403","GM":"0000"}
QoS: 0 - Retain: false
Nachricht 75 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
QoS: 0 - Retain: false
Nachricht 74 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
QoS: 0 - Retain: false
Nachricht 73 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 72 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024731313340425E03","G1":"13@B"}
QoS: 0 - Retain: false
Nachricht 71 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"02474D303030305403","GM":"0000"}
QoS: 0 - Retain: false
Nachricht 70 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
QoS: 0 - Retain: false
Nachricht 69 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
QoS: 0 - Retain: false
Nachricht 68 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 67 empfangen auf info/24EC4A3237D4/rx um 11:20:
{"protocol":"S21","dump":"024731313342426003","G1":"13BB"}
QoS: 0 - Retain: false
[... 98 lines cut ...]
QoS: 0 - Retain: false
Nachricht 49 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246377D03","F7":""}
QoS: 0 - Retain: false
Nachricht 48 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246367C03","F6":""}
QoS: 0 - Retain: false
Nachricht 47 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246357B03","F5":""}
QoS: 0 - Retain: false
Nachricht 46 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246317703","F1":""}
QoS: 0 - Retain: false
Nachricht 45 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"02464D9303","FM":""}
QoS: 0 - Retain: false
Nachricht 44 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246377D03","F7":""}
QoS: 0 - Retain: false
Nachricht 43 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246367C03","F6":""}
QoS: 0 - Retain: false
Nachricht 42 empfangen auf info/24EC4A3237D4/tx um 11:21:
{"protocol":"S21","dump":"0246357B03","F5":""}
QoS: 0 - Retain: false

revk, 2025-08-15:
OK so we see `G1` as `13@B` and `13BB` which are 18℃ and 19℃.

Is that what you tried.

Are you saying the Faikin did not update and show 18℃ and 19℃ at all?

Jxhannes, 2025-08-15:
Cooling mode is correct. I thought I had tried more temperature setpoints (started at 24 °C, then down and back up), but maybe not everything was logged.
Faikin and Home Assistant show the correct target temperature (18–19 °C in this case). The issue is with reading the indoor and outdoor temperatures.

[image]

borgqueenx, 2025-08-15:
On a offtopic question, I assume this cheaper daikin does not support liquid temperature sensor and kwh consumption tracking?

revk, 2025-08-15:
Oh, OK, I was looking at the temperature setting, grrr.

OK we need `G9` for *home*/*outside*.

I am not seeing any `G9` in your logs.

So, can you start the Faikin and right away get MQTT `rx` messages with `G9` in them, and `error` messages. Well any messages with `G9` in it, or `F9`... They should get sent just after startup at the very least.

Jxhannes, 2025-08-15:
5.08.2025 12:17:39(-0.5 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 12:17:34(-5.3 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 12:17:31(-2.3 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 12:17:31(-0.39 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 12:17:28(-2.79 seconds)
{"protocol":"S21","timeout":true}
15.08.2025 12:17:28(-0.48 seconds)
{"protocol":"S21","cmd":"FA","mismatch":true,"data":"024700000000004703"}
15.08.2025 12:16:08(-79.79 seconds)
{"protocol":"S21","badsum":"15","data":"024739B0B6FF301703"}
15.08.2025 12:15:32(-35.92 seconds)
{"protocol":"S21","badsum":"15","data":"02

Jxhannes, 2025-08-15:
and rx after activating all protocols, outdoor temp changed again 

Nachricht 15 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024700000000004703","G":"\u0000\u0000\u0000\u0000"}
QoS: 0 - Retain: false
Nachricht 14 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
QoS: 0 - Retain: false
Nachricht 13 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
QoS: 0 - Retain: false
Nachricht 12 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 11 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024734302080307B03","G4":"0 \u00800"}
QoS: 0 - Retain: false
Nachricht 10 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024733303030303A03","G3":"0000"}
QoS: 0 - Retain: false
Nachricht 9 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024732343A8080E703","G2":"4:\u0080\u0080"}
QoS: 0 - Retain: false
Nachricht 8 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"02473130334C416803","G1":"03LA"}
QoS: 0 - Retain: false
Nachricht 7 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024700000000004703","G":"\u0000\u0000\u0000\u0000"}
QoS: 0 - Retain: false
Nachricht 6 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
QoS: 0 - Retain: false
Nachricht 5 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
QoS: 0 - Retain: false
Nachricht 4 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024735303030303C03","G5":"0000"}
QoS: 0 - Retain: false
Nachricht 3 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024734302080307B03","G4":"0 \u00800"}
QoS: 0 - Retain: false
Nachricht 2 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024733303030303A03","G3":"0000"}
QoS: 0 - Retain: false
Nachricht 1 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"024732343A8080E703","G2":"4:\u0080\u0080"}
QoS: 0 - Retain: false
Nachricht 0 empfangen auf info/24EC4A3237D4/rx um 12:19:
{"protocol":"S21","dump":"02473130334C416803","G1":"03LA"}

Jxhannes, 2025-08-15:
okay, rx topic?

Jxhannes, 2025-08-15:
Is there any better way to save topics? HA and MQTT explorer both have a limited saving range. If I want to catch Ra/Rh I have to save somehow hundrets of messages from Rx because it gets spammed with messages

revk, 2025-08-15:
Can you not look and see what comes out... It is `tx` and `rx` by the look of it.

revk, 2025-08-15:
I just watch MQTT on a console, sorry. Let me look at a way to do the debug level summary without seeing extra commands...

alex-sever-h, 2025-08-15:
I tracked all the messages with mosquitto_sub in the console and Rh and RA was not present at all.

revk, 2025-08-15:
OK, I am making so `debug` just dumps the status, and sorry it dumps reply so `SH` and `Sa` not `RH` and `Ra`...

For example, I see

```
info/GuestAC/s21 {"G1":"03@A","G5":"000\u0080","G6":"000<","G7":"00@=","GM":"D5CA","SG":"A","SH":"012+"}
info/GuestAC/s21 {"G1":"03@A","G5":"000\u0080","G6":"000<","G7":"00@=","GM":"D5CA","SG":"A","SI":"561+"}
info/GuestAC/s21 {"G1":"03@A","G5":"000\u0080","G6":"000<","G7":"00@=","GM":"D5CA","SG":"A","Sa":"552+"}
info/GuestAC/s21 {"G1":"03@A","G5":"000\u0080","G6":"000<","G7":"00@=","GM":"D5CA","SG":"A","SL":"000"}
info/GuestAC/s21 {"G1":"03@A","G5":"000\u0080","G6":"000<","G7":"00@=","GM":"D5CA","SG":"A","Sd":"810"}
info/GuestAC/s21 {"G1":"03@A","G5":"000\u0080","G6":"000<","G7":"00@=","GM":"D5CA","SG":"A","SN":"500-"}
```

The `SH` and `Sa` are the *home* and *outside* temps (21.0℃ and 25.5℃).

So upgrade to beta release and turn on `debug` and log those `s21` topics.

Jxhannes, 2025-08-15:
Okay, I will update to beta and will try

revk, 2025-08-16:
Any chance of what I said, `debug` (not `dump`) and the `s21` topic.

The `tx` of `Ra` is simply the polling for the `Sa` response and tells me nothing.

Jxhannes, 2025-08-16:
Thsi is comms with activated debug 

16.08.2025 11:12:13(-0.99 seconds)
{"protocol":"S21","cmd":"Ra","nak":true}
16.08.2025 11:12:12(-1.01 seconds)
{"protocol":"S21","cmd":"RI","nak":true}
16.08.2025 11:12:11(-1 seconds)
{"protocol":"S21","cmd":"RH","nak":true}
16.08.2025 11:12:10(-1 seconds)
{"protocol":"S21","cmd":"RG","nak":true}
16.08.2025 11:12:09(-1 seconds)
{"protocol":"S21","cmd":"RN","nak":true}
16.08.2025 11:12:07(-2 seconds)
{"protocol":"S21","cmd":"RL","nak":true}
16.08.2025 11:12:06(-1 seconds)
{"protocol":"S21","cmd":"Ra","nak":true}
16.08.2025 11:12:05(-1 seconds)
{"protocol":"S21","cmd":"RI","nak":true}
16.08.2025 11:12:04(-1 seconds)
{"protocol":"S21","cmd":"RH","nak":true}
16.08.2025 11:12:03(-0.99 seconds)
{"protocol":"S21","cmd":"RG","nak":true}
16.08.2025 11:12:02(-1.01 seconds)
{"protocol":"S21","cmd":"RN","nak":true}
16.08.2025 11:12:00(-2 seconds)
{"protocol":"S21","cmd":"RL","nak":true}
16.08.2025 11:11:59(-0.98 seconds)
{"protocol":"S21","cmd":"Ra","nak":true}
16.08.2025 11:11:58(-1 seconds)
{"protocol":"S21","cmd":"RI","nak":true}
16.08.2025 11:11:56(-1.57 seconds)
{"protocol":"S21","cmd":"RH","nak":true}
16.08.2025 11:11:50(-5.62 seconds)
{"protocol":"S21","cmd":"RI","nak":true}
16.08.2025 11:11:49(-0.99 seconds)
{"protocol":"S21","cmd":"RH","nak":true}
16.08.2025 11:11:48(-1 seconds)
{"protocol":"S21","cmd":"RG","nak":true}

Jxhannes, 2025-08-16:
Okay thank you, I just installed it. I will report any news!

borgqueenx, 2025-08-16:
[image]

revk, 2025-08-16:
That's an S3-MINI-8, arg, I don't make them.

OK, they me see if I can sort a beta for that!

borgqueenx, 2025-08-16:
Thanks the upgrade worked. After updating the temperature shows on dashboard and home assistant, probably because of a reboot. However, after half an hour, no changes in temperature. I also started the ac and turned it off for 10 minutes. Still no differences in temeperatures.

revk, 2025-08-16:
I do wonder if those *nak* suggest a hardware/wiring issue.

revk, 2025-08-16:
OK who has the issue?

borgqueenx, 2025-08-16:
> OK who has the issue?

Both boards have the issue? Thats why @Jxhannes said its probably not a hardware error.

revk, 2025-08-16:
Answer may be a setting to force `F9` use, perhaps.

I'll look tomorro

revk, 2025-08-16:
Are you seeing the same *nak*s

Jxhannes, 2025-08-16:
I think yes:

16.08.2025 20:48:35(-3.99 seconds)
{"protocol":"S21","cmd":"Ra","nak":true}
16.08.2025 20:48:33(-2.02 seconds)
{"protocol":"S21","cmd":"RH","nak":true}
16.08.2025 20:48:31(-1.98 seconds)
{"protocol":"S21","cmd":"RN","nak":true}
16.08.2025 20:48:29(-2 seconds)
{"protocol":"S21","cmd":"RL","nak":true}
16.08.2025 20:48:27(-2 seconds)
{"protocol":"S21","cmd":"RI","nak":true}
16.08.2025 20:48:25(-2 seconds)
{"protocol":"S21","cmd":"RG","nak":true}
16.08.2025 20:48:21(-4 seconds)
{"protocol":"S21","cmd":"Ra","nak":true}
16.08.2025 20:48:19(-2 seconds)
{"protocol":"S21","cmd":"RH","nak":true}
16.08.2025 20:48:17(-2 seconds)
{"protocol":"S21","cmd":"RN","nak":true}
16.08.2025 20:48:15(-2 seconds)
{"protocol":"S21","cmd":"RL","nak":true}

revk, 2025-08-16:
Then I am at a bit of a loss. It have been very hard to progress any debugging on this, but getting those *nak* is odd.

revk, 2025-08-20:
It gets a reply at least once, but many NAKs, so gives up.

Maybe I could make it not give up, but that does not explain why it often NAKs.

tjz, 2026-01-05:
Perhaps the only reason it works on boot (or second boot as it happens for me often) is because it gets other commands before and expects some other command to have occurred at least x seconds before providing a response as to the internal/external temperature. Perhaps the original Daikin BRPs always send out specific commands in specific order, e.g. asking for all the temps in a burst of requests, not just specific temps... Or a full status of sorts... Just putting it out there because I've been having this issue for a long time.

Jxhannes, 2026-01-05:
Still makes no sense why it would work on every other model.  Why would it be different on  this specific indoor unit

endelwar, 2026-01-06:
I have the same issue on my 2020 Siesta (`atxc35bv1b`), running `Faikout-S3-MINI-N4-R2: 34cb9551 2026-01-06T07:45:56 S21`
I need to reboot the device to get the correct temperature reading.
Is there anything I can do to help move this issue forward?