revk/xESP32-Faikout #783: oscillation over MQTT not working properly if noswingh is enabled

PumpgunRonnie, 2025-07-27:
**Faikin hardware**
Purchased Faikin-S3-MINI-N8.

**Daikin hardware**
Daikin FTXS20K2V1B, connected to S403 Port.

**Describe the bug**
If noswingh is enabled i can activate the vertikal swing via the web overlay properly, via MQTT on HomeAssistant I can only turn in off and not on again. After 2-3 seconds it just shows off again. If noswingh is off i can activate the vertikal swing over MQTT without any problem. So must be a bug and the function would work properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate noswingh in Options -> Extra.
2. Try to activate vertikal swing in my HomeAssistant.
3. After some seconds and no movement at all it just shows off again.

**Expected behavior**
Activate swing in HomeAssistant via MQTT without showing the horizontal swing option which my A/C does not have.

revk, 2025-07-27:
More logs of messages, and where did you buy?

PumpgunRonnie, 2025-07-27:
MQTT Log from the AC after trying to activate swing:

error:

control = {
  "field": "swingh",
  "error": "Setting cannot be controlled"
}

swing = {
  "ts": "2025-07-27T15:18:39Z",
  "description": "Setting cannot be controlled",
  "prefix": "command",
  "suffix": "swing",
  "payload": "on"
}

"field": "swingh" seems for me like if noswingh is active, the MQTT command swing = on from my HomeAssistant means for the Faikin swingh should be turned on instead of swingv which gives back an error because it cannot control it.

If noswingh is off HomeAssistant sends the command "swing = V" instead of "swing = on" which works fine without an error.

And I buyed them from someone in an online forum who had more then he needed, i dont know where he got them.