revk/xESP32-Faikout #789: [FEATURE] Manual flap control (no swing)

blami, 2025-08-07:
First great job! I ordered 4 faikins for all my units. I needed to add optional board with S21 but that is fairly easy to get in BicCamera here in Japan and install (I have EE degree but going step by step manual included with board + following all safety precautions I believe its very easy process). Faikin works like a charm.

**Daikin hardware**
F22UTES-W (S22UTES-W set with external unit) + added KRP067A41 interface board with S21 lead

**Is your feature request related to a problem? Please describe.**
This is domestic Japanese "E-series" unit with horizontal only swing flap (vertical adjustment is manual by hand). For most Japanese units in addition to the swing there are multiple angle levels on remote control that put flap in that specific angle without swinging it. I think it would be nice to be able to customize a bit more how swing menu should look like (e.g. disable only vertical/horizontal mode and provide option to add/remove fixed angles). 

Let me know if any debug info is needed. I also can easily procure and donate all of: KRP067A41 (interface board that adds S21), remote and official wifi based remote BRP087A42 if that would be any help.

**Describe the solution you'd like**
I'd like to have fan-like control (HA is enough as that is what I use) where I could select flap angle. Eventually this can be added to swing mode. Like 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Note left middle part of remote display. It shows all angles and how this works is pressing **風向上下** button cycles through them from top-most to bottom-most and last option is horizontal swing. 
[image]

revk, 2025-08-07:
This is a bit of a stock reply for this, sorry.

The S21 is all reverse engineered and has a lot of options depending on the unit. Some units have features (that work with remote, etc) that are NOT exposed at all on S21. This means the feature cannot work over S21 at all.

Sometimes there is a feature and it is on S21, on some models. The key here is if the official Daikin WiFi / S21 module can do it then Faikin can do it - we just need to find the code to use. Indeed, this can be something that can be simulated or monitored from a real Daikin remote and the right AC.

There are ways (via MQTT command) to send S21 commands and get responses using the Faikin. This can be used (and some people have python code to play with it) to try different polls and commands to find new S21 codes.

If you, or someone else, finds an S21 poll/command, I am happy too add to Faikin code. But I cannot do much from here without the specific model you have.

Someone else may come along with more info on how to try these commands and play around.

As this comes up quite often, I am wondering if I can make more of a special debug page / interface for people somehow to *play* with S21 codes and find answers.

revk, 2025-08-07:
OK there is a `command/Faikin/send`, should be documented, and it says the reply, but worth reading some of the discussions to see what people have found and suggestions.

I am basically taking a bit of a back seat on that - tell me when there is a command people have found, and I'll be on it adding to Faikin and linking with HA.