revk/xESP32-Faikout #790: Daikin Firmware upgrade to 2.0 - safe?

christianmue1980a, 2025-08-08:
Hi,

my Daikin Onecta App signals that new firmware updates for my Perfera / Emura devices are available. From 1_39_0 to 2_0_0. Is it generally safe to install firmware updates like these or are problems expected when Faikin is used as well?

Thanks in advance!

revk, 2025-08-08:
I doubt it would be an issue.

But see what other people say.

zaubara, 2025-08-13:
I don't see any difference to how it worked before, so it's probably safe.

dahadd, 2025-11-15:
Yes its safe

uscmbrixton, 2026-01-02:
A reason not to could be that it prevents you from controlling your unit locally with for example Home Assistant. I have not verified this myself but is seems this integration worked with a previous firmware but no longer with 2.0, then your only option seem to be cloud or Faikin.