revk/xESP32-Faikout #799: [FEATURE] Expose BLE advertisements

solstead, 2025-08-20:
Would love to have a setting to expose BLE advertisements for use with a presence tracking integration like Bermuda.

revk, 2025-08-20:
Err, hmm, intersting

skatealaska, 2026-01-05:
Seconding - I'd love to have the BLE proxy available for the same reason (to use Bermuda).  It would be super useful for me because members of the household frequently leave mini-splits on when they've left the room, but not everyone wants them on always when they are there (so it lets you distinguish individual family presences) and would improve the triangulation accuracy for all 4 rooms (since 2 of the rooms currently only have an under-bed sensor, and the other two use a single router on their shared wall)