revk/xESP32-Faikout #800: System is offline (no data tx/rx) Problem

bma45, 2025-08-21:
I bought the board on Amazon and connected it via S21 to my Daikin Stylish. Faikin also starts, but it shows "System is offline (no data tx/rx)".
I have already checked the cable multiple times.
What could be the issue?

revk, 2025-08-21:
Usually either wrong wiring or a fault in the board. We do check boards before shipping.

If you loop back the tx/rx to the Faikin, i.e. Faikin tx back to Faikin rx and not connected to the AC, that should say that it is in "loopback" instead of off line.

If that works, the Faikin is OK.

DanJoeStone, 2025-08-28:
Hi, do I need the 12V supplied to test loop back? Or is 5V pin connection good enough to test loop back?

revk, 2025-08-29:
You need power, i.e. on the 12V pin (can be 5V to 17V).

The 5V pin is not for powering the Faikin, and should not be needed to test loopback (unless a very old board).

skatealaska, 2025-10-22:
I had this issue and resolved it - it first appeared after a power surge at our house, then disappeared, then another power cut led to it stuck in this state.  The Faikin still worked and showed up with the web interface, the issue was actually a fried board in the unit itself from the power surges.  The technicians were able to replace the fried board and it all works again now.

skatealaska, 2025-10-24:
Yes!  The breaker fixed it the first time, the second it couldn’t, and the situation actually recurred recently and the breaker off wait 20s to 20m and back on did the fix!

raro71, 2026-01-17:
@revk  Thanks for the description. I have an RX/TX problem. I tried the loop diagnostics, but it failed – no problem: a faulty device, so what. I ordered a new one – but I still can't test the loop!???

12V, 5V, and GND are checked and fine. Why isn't there a loop signal?

I try to to link a FanCoil FWT02CATNMV1 at CN-WIR port. But didn´t get the rx/tx AND I´m not able to work out the loop test (twice)

THX for help

raro71, 2026-01-17:
You mean those one:

RX and TX with GND an 3V?

https://github.com/revk/ESP32-Faikout/blob/main/PCB/Faikout/Faikout-bottom.png

revk, 2026-01-17:
Latest beta should confirm loop back on web interface now

raro71, 2026-01-18:
I've now figured it out - both boards are ok (except the web interface bug - LEDs signals are proper)

Don´t know if there´s a hint in the setup section (I didn´t find one), but it seems to be important to connect the RX/TX and GND first, leave the 5V line unconnected, turn on the fancoil (only electricity not running) and then put on the 12V. During the few seconds of booting the fancoil interface was detected and it works instantly and perfectly.

Might be an issue, that the fancoil needs more time "to start" the interface, than the module scans the interface. And after the scan, the module says: no interface ?!??

Regards and thx to @revk for the module!

revk, 2026-01-18:
Weird, maybe the auto detect logic is confusing it somehow. It may be possible to lock the settings down now for future.