revk/xESP32-Faikout #803: [FEATURE]Daikin FTXPxxN

Lampieyyy, 2025-08-24:
Hi, ich wollte nachfragen ob ich eine Daikin FTXPxxNxxxxx mit einem Faikin Adapter nutzen kann?
Hat das bereits jemand gelöst und kann Hilfestellung geben und Bilder schicken, wie die jeweilige Belegung ist?

revk, 2025-08-26:
Beware X403 is usually *live*, so Faikin will be *live* when connected.

Lampieyyy, 2025-08-26:
what that means?

I would connect as follows: 
S403          to S21 
Pin 1 (14V) to Pin 4, 
Pin 4 (TX) to Pin 2, 
Pin 5 (RX) to Pin 3, 
Pin 6 (GND) to Pin 5. 

Would this work like that, or what would be the next approach?
Should I still consider anything else?

Lampieyyy, 2025-08-29:
Initial situation: I have a Daikin FTXPxxNxxxxx and I wanted to integrate it into Home Assistant (HA). Description: For this I bought a Faikin module from Amazon and connected it to the S403 interface using individual jumpers. 
The wiring was as follows: 
Faikin – S403 
GND – GND pin 6 or 7, 
12V – +14V pin 1 (BOTTOM), 
48 (TX) – RX pin 5, 
34 (RX) – TX pin 4, 
5V – Not Used. 
Integrated in Home Assistant, I connected it via MQTT. It works flawlessly.