revk/xESP32-Faikout #809: [FEATURE] Daikin FTXC-E series wiring / S_DGC port wiring for Faikin.

redecs, 2025-09-02:
Hello,

Between me ordering two new AC units series FTXC-D (Daikin Sensira Bluevolution FTXC20D-RXC20D as described by my local vendor) the stock ran out and they gracefully replaced the model I ordered with it's next version the FTXC20E-RXC20E, that comes with included WLAN adapter.

**Daikin hardware**
Daikin Sensira Bluevolution FTXC20E-RXC20E

**Is your feature request related to a problem? Please describe.**
I would like to replace build WLAN adapter with a Faikin adapter so I can have complete local access to my AC units, without any cloud services.

**Describe the solution you'd like**
I would like to know if anyone managed to adapt a Faikin adapter to work with this model. The included WLAN adapter connects to a port labeled `S_DGC` that I have not seen mentioned in any documentation here.

**Describe alternatives you've considered**
I really like the Faikin approach, but if no luck I'll have to use an IR controller for my local automations without cloud services.

**Additional context**
Here's the diagram of device, provided by Daikin

[image]

I also took the covers off to look at the motherboard myself and there is a slot on the board for S21 but there's no header soldered there and I don't know if the pins are actually connected to anything. Here are some images:

[image]
[image]

I would really really like to avoid an IR controller, so any hints and advices regarding the S_DGC port or similar unites is greatly appreciated. 

Thanks!

kcsoft, 2025-11-28:
@redecs 

I have the same model, same builtin wifi module.

That `S_DGC`  connector is a 5pin GH1.25 pitch connector. I did manage to find exact one on aliexpress.
Since it is next to the `S21` and the traces seem to be common with the `S21`, I was hopping it is just a different header but using the same protocol.

I did try to put together a Faikout board from an "ESP32 C3 Super mini", a DC to DC MINI560 board and a 2k2+4k7 resistor divider for the ESP32 RX input.
I was using GPIO20 for RX (with the voltage divider) and GPIO21 for TX. I did set these in the Faikin settings.

I plugged the board and powers just fine, I was able to connect to the web page of the board and adjust the settings, connect to my MQTT server.

But the communication with the Daikin unit fails.

All I could see if the board failing to detect any sign of life on the rx/tx lines by timing out on different protocols:
```
error/FaikinBB/comms {"protocol":"S21¬Tx","timeout":true}
error/FaikinBB/comms {"protocol":"S21¬Rx","timeout":true}
error/FaikinBB/comms {"protocol":"S21¬Tx¬Rx","timeout":true}
error/FaikinBB/comms {"protocol":"X50A","timeout":true}
error/FaikinBB/comms {"protocol":"X50A¬Tx","timeout":true}
error/FaikinBB/comms {"protocol":"X50A¬Rx","timeout":true}
error/FaikinBB/comms {"protocol":"X50A¬Tx¬Rx","timeout":true}
error/FaikinBB/comms {"protocol":"S21","timeout":true}
error/FaikinBB/comms {"protocol":"S21¬Tx","timeout":true}
error/FaikinBB/comms {"protocol":"S21¬Rx","timeout":true}
error/FaikinBB/comms {"protocol":"S21¬Tx¬Rx","timeout":true}
```

I did try different protocols/settings but all failed.

I couldn't really hardware debug on it because the physical access is tight and I can just barely insert the connector in. 

[image]

redecs, 2025-11-28:
Thanks for update @kcsoft. Since the cold weather came, I haven't had any time to fiddle with my AC unit.

ramonv, 2025-12-12:
following