revk/xESP32-Faikout #810: [BUG] val_tpl for AC Target causes HA warnings when actarget is missing

eupesco, 2025-09-02:
I am on the latest beta.

The `...ac_target` sensor is published with :
{
  "name": "AC-Target",
  "state_topic": "state/.....AC",
  "unit_of_meas": "°C",
  "dev_cla": "temperature",
  "state_class": "measurement",
  **"val_tpl": "{{value_json.actarget}}",**
  ...
}

HA logs warnings whenever the state JSON does not include actarget (e.g. when AC is off):
```
Logger: homeassistant.helpers.template
Quelle: helpers/template.py:2990
Template variable warning: 'dict object' has no attribute 'actarget' when rendering '{{value_json.actarget}}'
```

Suggestion:
Update `val_tpl` to a safe default when AC is off (so when actarget is missing)`` 
For example:
`"val_tpl": "{{ (value_json.actarget | default(0)) | float | round(1) }}"`

revk, 2025-09-02:
When do we send only availability (apart from when power is off to the Faikin as a will)?

eupesco, 2025-09-02:
I checked again with MQTT explorer:

```
"stat_t": "state/****AC",
"val_tpl": "{{value_json.actarget}}"
```
But the actual payloads on state/****AC do not contain actarget.
There is target, for example:

`{"target": [19.5, 20.5], "online": true, ... }`

Since the discovery config expects actarget, HA logs a warning whenever the payload only has target.

eupesco, 2025-09-02:
Thanks for clarifying. You are right, the example with` "target": [19.5,20.5]` was from `Faikin/****AC`. 
On `state/****AC` I do see payloads without actarget. Then HA logs the template warning.

revk, 2025-09-02:
OK so why is HA doing that?

eupesco, 2025-09-02:
Makes sense. Looks like my broker still had an old retained state from the first power-up without actarget.
After toggling the ACs once, everything is fine now. I’ll try to reproduce it later.