revk/xESP32-Faikout #813: ESP32 C3 RX/TX pinout

Cliff-R-K, 2025-09-03:
I have a ESP32 C3 Super Mini and have flashed the C3 firmware. It boots normaly and I can log in to the WebUI. But it shows System Offline whenever I connect it to my Daikin.
From reading some posts here it looks like I also need a logic level shifter 5->3.3v to get it to work, and I have ordered one.

But I also noticed that when I look at the Extra-section in the UI it says Tx: -26 Rx: -27. If I look at the pinout for the C3 (https://forum.arduino.cc/t/esp32-c3-supermini-pinout/1189850) the Tx-pin is 21 and Rx 20.

Do I also need to change this ? and should I change Tx to -21 and Rx to -20 or just 21 and 20 ?

revk, 2025-09-03:
You set whichever GPIO pins you want to use. The - inverts the logic but the code tried different combinations of inverts by default anyway. They are inverted on the Faikin because the level change logic inverts.

Cliff-R-K, 2025-09-03:
Thanks, do you also know if it could be possible to connect any esp32 without a level shifter or is this needed for all variants ?

revk, 2025-09-03:
Well it may work or it may break. The Daikin is typically 5V signalling and the ESP is 3.3V.

The issues are (a) 5V may damage ESP, and (b) 3.3V may not be enough for Daikin to *see* it.