revk/xESP32-Faikout #822: [FEATURE] Support retained commands

timbuchwaldt, 2025-09-20:
**Is your feature request related to a problem? Please describe.**
I'm trying to optimize my power consumption by turning my whole AC system off with a timer. For that I have some logic in my central controller that waits for 1h after the last commanded heating/cooling and then turns off power to the whole AC system. When heating/cooling is commanded again, it will power up the circuit again.
Setting the MQTT commands happens asynchronous to this due to the inner workings of Loxone (the control system) - ergo I'm commanding via MQTT while the Faikin modules are potentially still off / booting / connecting.

**Describe the solution you'd like**
I'd love for retained command messages to work, so I can asynchronously send my commands and when Faikin connects to my MQTT broker it would be in my desired state.

**Describe alternatives you've considered**
Two hacky ones:
- Somehow delay sending my commands in case my central power switch is off
- Having a tool in the middle that repeats my commands based on the last commanded state.

revk, 2025-09-20:
Well tricky, as down to MQTT server to send the message. The Faikin does not care if it is marked retained or not.

revk, 2025-09-21:
Ah, that is possible, which makes it a bit tricky... We don't currently delay. It was not really designs to be turned off, to be honest - does your AC really use much when simply "switched off".