revk/xESP32-Faikout #823: FBA71A2VEB9 ceiling unit

fonzy69, 2025-09-22:
Hi all,

i just get a new installation with FBA71A2VEB9 ceiling unit. It comes with a brp069c81 remote that works only with daikin's cloud. 
The unit is supposed to work with older brp069a81 with local access but this one is not available (too old)
Here is some pictures

[image]

[image]

It seems to be X35 + X504A connectors but with only two wires on X540A

Do you think it will be working with Faikin and what kind of wiring does i need?

Thanks for your help.

revk, 2025-09-22:
4 wires is usually OK, two will be GND/12V and two will be Tx/Rx, typically. The X35 is power, normally.

The other two look like it is labeled X50A not X504A/X540A which should indeed work.

There is a lot of info on the wiring on https://github.com/revk/ESP32-Faikin/wiki/Wiring
If you have a lead to an S21/5 way connector that should just plug in.

fonzy69, 2025-09-24:
Once again you are right...
do you know what is the original purpose of the unconnected wire?
So i could plug the original cable that go to the brp069c81 in the faikin module?
Thanks for your help

closed 2025-09-24 by revk