revk/xESP32-Faikout #833: [BUG] Faikin-auto Mode Target slider not working

bjappler, 2025-09-27:
Hi,

I noticed a strange behaviour. In one of my 12 Faikin Modules it is no longer possible to control the target temperature in the front end (no + or - or slider ist working. It is still possible to change temperature in the backend or with mqtt.
I tried to update Faikin to the newest Beta but the problem still remains. In my others ACs Faikins there is not a problem. I also restarted the AC and Faikin Module

What can I do?
In my others ACs Faikins there is not a problem.

[image]

revk, 2025-09-27:
Wow, no clue - does the browser debug show any errors in javascript, etc?

The set at 10C is also oddly low.

bjappler, 2025-09-27:
[image]

Set temperature of 10C was correct, as AC was not heating this moment.